CZ-NIC/turris-os #1: systemd?

zabbal, 2015-11-12:
Any chance to see ideas from https://github.com/thagabe/openwrt-systemd implemented in turris?
The closer OS in my router to modern GNU/Linux the better.

samm-git, 2015-11-17:
@zabbal procd in openwrt do it job pretty well, i dont see any real value in replacing it. However, you can easily build this feed and play yourself, i dont think there is anything turris-os specific here

sairon, 2015-11-17:
Closing this as "wontfix". There are several problems - for example the need for glibc or divergence from upstream - by replacing procd with systemd, you'll basically have to rewrite all the services and init scripts. There's really nothing reasonable about this transition.

closed 2015-11-17 by sairon