CZ-NIC/turris-os #2: Compiling for AR7xxx/AR9xxx fails

DatanoiseTV, 2015-11-23:
I am currently trying to adapt the Turris OS to run on AR7xxx/AR9xxx platforms, such as the MikroTik RouterBoard 450G, but the build process silently fails. The R450G seems like a nice platform to be supported by Turris. My next platform to be ported to, will be an Octeon based system, by the way, if there is any interest.

```
mips-openwrt-linux-uclibc-objcopy -O binary -R .reginfo -R .notes -R .note -R .comment -R .mdebug -R .note.gnu.build-id -S /usr/src/turris-os/build_dir/target-mips_34kc_uClibc-0.9.33.2/linux-ar71xx_mikrotik/linux-3.18.21/vmlinux /usr/src/turris-os/build_dir/target-mips_34kc_uClibc-0.9.33.2/linux-ar71xx_mikrotik/vmlinux
mips-openwrt-linux-uclibc-objcopy -R .reginfo -R .notes -R .note -R .comment -R .mdebug -R .note.gnu.build-id -S /usr/src/turris-os/build_dir/target-mips_34kc_uClibc-0.9.33.2/linux-ar71xx_mikrotik/linux-3.18.21/vmlinux /usr/src/turris-os/build_dir/target-mips_34kc_uClibc-0.9.33.2/linux-ar71xx_mikrotik/vmlinux.elf
cp -fpR /usr/src/turris-os/build_dir/target-mips_34kc_uClibc-0.9.33.2/linux-ar71xx_mikrotik/linux-3.18.21/vmlinux /usr/src/turris-os/build_dir/target-mips_34kc_uClibc-0.9.33.2/linux-ar71xx_mikrotik/vmlinux.debug
mv /usr/src/turris-os/build_dir/target-mips_34kc_uClibc-0.9.33.2/linux-ar71xx_mikrotik/linux-3.18.21/.config /usr/src/turris-os/build_dir/target-mips_34kc_uClibc-0.9.33.2/linux-ar71xx_mikrotik/linux-3.18.21/.config.old
grep -v -e INITRAMFS -e CONFIG_RD_ -e CONFIG_BLK_DEV_INITRD /usr/src/turris-os/build_dir/target-mips_34kc_uClibc-0.9.33.2/linux-ar71xx_mikrotik/linux-3.18.21/.config.old > /usr/src/turris-os/build_dir/target-mips_34kc_uClibc-0.9.33.2/linux-ar71xx_mikrotik/linux-3.18.21/.config
echo 'CONFIG_BLK_DEV_INITRD=y' >> /usr/src/turris-os/build_dir/target-mips_34kc_uClibc-0.9.33.2/linux-ar71xx_mikrotik/linux-3.18.21/.config
echo 'CONFIG_INITRAMFS_SOURCE="/usr/src/turris-os/build_dir/target-mips_34kc_uClibc-0.9.33.2/root-ar71xx /usr/src/turris-os/target/linux/generic/image/initramfs-base-files.txt"' >> /usr/src/turris-os/build_dir/target-mips_34kc_uClibc-0.9.33.2/linux-ar71xx_mikrotik/linux-3.18.21/.config
echo 'CONFIG_INITRAMFS_ROOT_UID=1000' >> /usr/src/turris-os/build_dir/target-mips_34kc_uClibc-0.9.33.2/linux-ar71xx_mikrotik/linux-3.18.21/.config
echo 'CONFIG_INITRAMFS_ROOT_GID=1000' >> /usr/src/turris-os/build_dir/target-mips_34kc_uClibc-0.9.33.2/linux-ar71xx_mikrotik/linux-3.18.21/.config
echo "CONFIG_INITRAMFS_COMPRESSION_NONE=y" >> /usr/src/turris-os/build_dir/target-mips_34kc_uClibc-0.9.33.2/linux-ar71xx_mikrotik/linux-3.18.21/.config
echo -e "# CONFIG_INITRAMFS_COMPRESSION_GZIP is not set\n# CONFIG_RD_GZIP is not set" >> /usr/src/turris-os/build_dir/target-mips_34kc_uClibc-0.9.33.2/linux-ar71xx_mikrotik/linux-3.18.21/.config
echo -e "# CONFIG_INITRAMFS_COMPRESSION_BZIP2 is not set\n# CONFIG_RD_BZIP2 is not set" >> /usr/src/turris-os/build_dir/target-mips_34kc_uClibc-0.9.33.2/linux-ar71xx_mikrotik/linux-3.18.21/.config
echo -e "# CONFIG_INITRAMFS_COMPRESSION_LZMA is not set\n# CONFIG_RD_LZMA is not set" >> /usr/src/turris-os/build_dir/target-mips_34kc_uClibc-0.9.33.2/linux-ar71xx_mikrotik/linux-3.18.21/.config
echo -e "# CONFIG_INITRAMFS_COMPRESSION_LZO is not set\n# CONFIG_RD_LZO is not set" >> /usr/src/turris-os/build_dir/target-mips_34kc_uClibc-0.9.33.2/linux-ar71xx_mikrotik/linux-3.18.21/.config
echo -e "# CONFIG_INITRAMFS_COMPRESSION_XZ is not set\n# CONFIG_RD_XZ is not set" >> /usr/src/turris-os/build_dir/target-mips_34kc_uClibc-0.9.33.2/linux-ar71xx_mikrotik/linux-3.18.21/.config
echo -e "# CONFIG_INITRAMFS_COMPRESSION_LZ4 is not set\n# CONFIG_RD_LZ4 is not set" >> /usr/src/turris-os/build_dir/target-mips_34kc_uClibc-0.9.33.2/linux-ar71xx_mikrotik/linux-3.18.21/.config
cp -fpR /usr/src/turris-os/target/linux/generic/base-files/init /usr/src/turris-os/build_dir/target-mips_34kc_uClibc-0.9.33.2/root-ar71xx/init
cp: cannot create regular file `/usr/src/turris-os/build_dir/target-mips_34kc_uClibc-0.9.33.2/root-ar71xx/init': No such file or directory
make[4]: *** [/usr/src/turris-os/build_dir/target-mips_34kc_uClibc-0.9.33.2/linux-ar71xx_mikrotik/linux-3.18.21/.image] Error 1
make[4]: Leaving directory `/usr/src/turris-os/target/linux/ar71xx'
make[3]: *** [compile] Error 2
make[3]: Leaving directory `/usr/src/turris-os/target/linux'
make[2]: *** [target/linux/compile] Error 2
make[2]: Leaving directory `/usr/src/turris-os'
make[1]: *** [/usr/src/turris-os/staging_dir/target-mips_34kc_uClibc-0.9.33.2/stamp/.target_compile] Error 2
make[1]: Leaving directory `/usr/src/turris-os'
make: *** [world] Error 2
````

samm-git, 2015-11-26:
Why dont you want to use stock openwrt? Turris OS is based on it + some turris hw specific changes

MartinStrbacka, 2015-12-02:
Sorry guys we have no plans to support other HW than ours. If we will change our minds I let you know.

closed 2015-12-02 by MartinStrbacka