CZ-NIC/turris-os #3: turris-next shows tonns of flash related errors on the console

samm-git, 2015-11-26:
I installed turris-next on the testing router and found that it works mostly fine, but there are tonn of ecc errors (related to NAND) during boot and file writes. It looks this way:

```
[   91.037665] fsl-lbc ffe05000.localbus: Parity or Uncorrectable ECC error: LTESR 0x20000001
[   91.046317] fsl-lbc ffe05000.localbus: Parity or Uncorrectable ECC error: LTESR 0x20000001
[   91.074728] fsl-lbc ffe05000.localbus: Parity or Uncorrectable ECC error: LTESR 0x20000001
```
Or like that:
```
[    8.547064] fsl-lbc ffe05000.localbus: Parity or Uncorrectable ECC error: LTESR 0x20000001
[    8.555343] fsl,elbc-fcm-nand ff800000.nand: command failed: fir 4125e000 fcr 300000 status 20000001 mdr 0
[    8.565754] jffs2: mtd->read(0x9d4 bytes from 0x1180700) returned ECC error
[    8.581588] fsl-lbc ffe05000.localbus: Parity or Uncorrectable ECC error: LTESR 0x20000001
```

Despite this errors system seems to work fine. I am using JFFS version, please let me know if you need any additional information.

MartinStrbacka, 2015-12-02:
Samm-git probably forgot to run "nand scrub.chip -y" before flashing new kernel.

samm-git, 2015-12-02:
@MartinStrbacka  thank you for reply. No, it is something different. Now i did `nand scrub.chip -y` and re-created all filesystem but after reboot i still have tonn of the ECC errors in the console. I will also try tomorrow to restore from medkit, just to be 1000% sure, but i think it would not help

sairon, 2016-05-30:
The culprit was probably a worn out NAND. Can be resolved by better ECC on UBIFS.

closed 2016-05-30 by sairon