CZ-NIC/turris-os #5: kernel modules missing in kmod-crypto-iv and kmod-crypto-rng

ChristianKuehnel, 2016-10-08:
In the packages  kmod-crypto-iv and kmod-crypto-rng the kernel modules are missing. The packages contains only the scripts to load the modules. See [3] for a longer discussion.

The make script [1] looks fine to me, but in the repository [2] the packages are broken.

[1] https://github.com/CZ-NIC/turris-os/blob/master/package/kernel/linux/modules/crypto.mk
[2] https://api.turris.cz/openwrt-repo/omnia/packages/base/
[3] https://discourse.labs.nic.cz/t/strongswan-error-unable-to-install-inbound-and-outbound-ipsec-sa-sad-in-kernel/987/9

ChristianKuehnel, 2016-10-08:
Found a solution: installing the package strongswan-mod-kernel-libipsec provides the missing functionality

https://discourse.labs.nic.cz/t/strongswan-error-unable-to-install-inbound-and-outbound-ipsec-sa-sad-in-kernel/987/18

achim71, 2016-10-09:
Also getting errors when i test strongswan with an config which works on an different router running caos calmer (Kernel 3.18.23)
```
received netlink error: No such file or directory (2)
unable to add SAD entry with SPI cf94fd61
```
I think the kernel packages crypro module config needs this patch https://github.com/openwrt-mirror/openwrt/commit/5d40955aea8690ffcbdf57a7d7bfe61ef8bc11df

```

    kernel/modules: fix crypto API RNG for >=4.2

    Since kernel 4.2, DRBG is the default crypto API RNG, replacing krng. As
    DRBG is not enabled, there is no crypto API RNG available when running
    kernel 4.2 or later. Because of this, IPsec SAs fail to install. In
    strongSwan, this results in a vague error that is difficult to debug:

    received netlink error: No such file or directory (2)

    Solve this by adding DRBG to the kmod-crypto-rng package. As enabling
    DRBG in the kernel config also enables the Jitterentropy RNG, include it
    in kmod-crypto-rng instead of having it in a separate package.

    Signed-off-by: Stijn Tintel stijn@linux-ipv6.be

    git-svn-id: svn://svn.openwrt.org/openwrt/trunk@47827 3c298f89-4303-0410-b956-a3cf2f4a3e73

```

ChristianKuehnel, 2016-10-09:
reopened as requested by @achim71

miska, 2016-10-18:
Cherry-picked, will be part of the next release.

achim71, 2016-10-19:
This is an excerpt from crypto.mk from openwrt master. As you can seee the crypto-iv package needs modifications, currently the module echainiv.ko is not build and included which results in above error
`received netlink error: No such file or directory (2)`

With below settings strongswan finaly works here.

```
define KernelPackage/crypto-rng
  TITLE:=CryptoAPI random number generation
  DEPENDS:=+kmod-crypto-hash +kmod-crypto-hmac +kmod-crypto-sha256
  KCONFIG:= \
	CONFIG_CRYPTO_DRBG \
	CONFIG_CRYPTO_DRBG_HMAC=y \
	CONFIG_CRYPTO_DRBG_HASH=n \
	CONFIG_CRYPTO_DRBG_MENU \
	CONFIG_CRYPTO_JITTERENTROPY \
	CONFIG_CRYPTO_RNG2
  FILES:= \
	$(LINUX_DIR)/crypto/drbg.ko@ge4.2 \
	$(LINUX_DIR)/crypto/jitterentropy_rng.ko@ge4.2 \
	$(LINUX_DIR)/crypto/krng.ko@lt4.2 \
	$(LINUX_DIR)/crypto/rng.ko
  AUTOLOAD:=$(call AutoLoad,09,drbg@ge4.2 jitterentropy_rng@ge4.2 krng@lt4.2 rng)
  $(call AddDepends/crypto)
endef

$(eval $(call KernelPackage,crypto-rng))

define KernelPackage/crypto-iv
  TITLE:=CryptoAPI initialization vectors
  DEPENDS:=+kmod-crypto-manager +kmod-crypto-rng +kmod-crypto-wq
  KCONFIG:= CONFIG_CRYPTO_BLKCIPHER2 CONFIG_CRYPTO_ECHAINIV
  FILES:= \
	$(LINUX_DIR)/crypto/eseqiv.ko \
	$(LINUX_DIR)/crypto/chainiv.ko \
	$(LINUX_DIR)/crypto/echainiv.ko@ge4.3
  AUTOLOAD:=$(call AutoLoad,10,eseqiv chainiv echainiv@ge4.3)
  $(call AddDepends/crypto)
endef

$(eval $(call KernelPackage,crypto-iv))

define KernelPackage/crypto-echainiv
  TITLE:=Encrypted Chain IV Generator
  DEPENDS:=+kmod-crypto-aead
  KCONFIG:=CONFIG_CRYPTO_ECHAINIV
  FILES:=$(LINUX_DIR)/crypto/echainiv.ko
  AUTOLOAD:=$(call AutoLoad,09,echainiv)
  $(call AddDepends/crypto)
endef

$(eval $(call KernelPackage,crypto-echainiv))

define KernelPackage/crypto-seqiv
  TITLE:=CryptoAPI Sequence Number IV Generator
  DEPENDS:=+kmod-crypto-aead +kmod-crypto-rng
  KCONFIG:=CONFIG_CRYPTO_SEQIV
  FILES:=$(LINUX_DIR)/crypto/seqiv.ko
  AUTOLOAD:=$(call AutoLoad,09,seqiv)
  $(call AddDepends/crypto)
endef

$(eval $(call KernelPackage,crypto-seqiv))

```

achim71, 2016-10-20:
Thanks for the patch. It should work. With 4.4 only echainiv.ko and seqiv.ko are build.
echainiv.ko was introduced with version 4.2 https://github.com/torvalds/linux/blob/v4.2/crypto/Makefile.
The modules chainiv.ko and esequiv.ko are no longer separate modules with 4.4 so the crypto-iv package does not contain any modules. I guess openwrt added the echainiv.ko module to that package for backward compatibility.
Thank you for your support.

achim71, 2016-10-28:
Tested the latest patches and can confirm strongswan is working with these. From my point of view the issue can be closed.

ChristianKuehnel, 2016-10-28:
closing as requested by @achim71

closed 2016-10-28 by ChristianKuehnel

rmayr, 2017-03-06:
With omnia-stable as of today, I had to manually install the package kmod-crypto-echainiv (with strongswan-full being installed) to make it work with the in-kernel ipsec stack. I suggest adding this dependency to strongswan-minimal.