CZ-NIC/turris-os #6: Buildenv failure in boot/uboot-envtool

achim71, 2016-10-09:
Hello,
I try to get an buld environment for the omnia-stable branch working on debian jessie. The make process fails on package boot/uenvtool. There have not been any recent changes on that package, so i'm abit lost what could cause that error.

```
make[3]: Entering directory '/root/turris/turris-os/package/boot/uboot-envtools'
make -C /root/turris/turris-os/build_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/u-boot-2014.10 CROSS_COMPILE="arm-openwrt-linux-uclibcgnueabi-" TARGET_CFLAGS="-Os -pipe -march=armv7-a -mtune=cortex-a9 -mfpu=vfpv3-d16 -fno-caller-saves -fhonour-copts -Wno-error=unused-but-set-variable -mfloat-abi=soft -I/root/turris/turris-os/staging_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/usr/include" UBI="" env
make[4]: Entering directory '/root/turris/turris-os/build_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/u-boot-2014.10'
scripts/kconfig/conf --silentoldconfig Kconfig
***
*** Configuration file ".config" not found!
***
*** Please run some configurator (e.g. "make oldconfig" or
*** "make menuconfig" or "make xconfig").
***
/root/turris/turris-os/build_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/u-boot-2014.10/scripts/kconfig/Makefile:36: recipe for target 'silentoldconfig' failed
make[6]: *** [silentoldconfig] Error 1
/root/turris/turris-os/build_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/u-boot-2014.10/Makefile:464: recipe for target 'silentoldconfig' failed
make[5]: *** [silentoldconfig] Error 2
  HOSTCC  tools/env/fw_env.o
tools/env/fw_env.c:56:2: error: unknown type name 'uint8_t'
  uint8_t mtd_type;  /* type of the MTD device */
  ^
tools/env/fw_env.c:81:2: error: unknown type name 'uint32_t'
  uint32_t crc; /* CRC32 over data bytes    */
  ^
tools/env/fw_env.c:86:2: error: unknown type name 'uint32_t'
  uint32_t crc; /* CRC32 over data bytes    */
  ^
tools/env/fw_env.c:99:2: error: unknown type name 'uint32_t'
  uint32_t  *crc;
  ^
tools/env/fw_env.c:111:1: error: unknown type name 'uint8_t'
 static uint8_t aes_key[AES_KEY_LENGTH] = { 0 };
 ^
tools/env/fw_env.c: In function 'fw_env_close':
tools/env/fw_env.c:342:31: error: 'uint8_t' undeclared (first use in this function)
  *environment.crc = crc32(0, (uint8_t *) environment.data, ENV_SIZE);
                               ^
tools/env/fw_env.c:342:31: note: each undeclared identifier is reported only once for each function it appears in
tools/env/fw_env.c:342:40: error: expected expression before ')' token
  *environment.crc = crc32(0, (uint8_t *) environment.data, ENV_SIZE);
                                        ^
tools/env/fw_env.c:342:40: error: too few arguments to function 'crc32'
In file included from tools/env/fw_env.c:36:0:
tools/env/fw_env.h:63:23: note: declared here
 extern unsigned long  crc32  (unsigned long, const unsigned char *, unsigned);
                       ^
tools/env/fw_env.c: At top level:
tools/env/fw_env.c:680:37: error: unknown type name 'uint8_t'
 static int flash_bad_block (int fd, uint8_t mtd_type, loff_t *blockstart)
                                     ^
tools/env/fw_env.c:708:21: error: unknown type name 'uint8_t'
       off_t offset, uint8_t mtd_type)
                     ^
tools/env/fw_env.c:797:22: error: unknown type name 'uint8_t'
        off_t offset, uint8_t mtd_type)
                      ^
tools/env/fw_env.c: In function 'env_aes_cbc_crypt':
tools/env/fw_env.c:1029:2: error: unknown type name 'uint8_t'
  uint8_t *data = (uint8_t *)payload;
  ^
tools/env/fw_env.c:1029:19: error: 'uint8_t' undeclared (first use in this function)
  uint8_t *data = (uint8_t *)payload;
                   ^
tools/env/fw_env.c:1029:28: error: expected expression before ')' token
  uint8_t *data = (uint8_t *)payload;
                            ^
tools/env/fw_env.c:1031:10: error: expected ';' before 'key_exp'
  uint8_t key_exp[AES_EXPAND_KEY_LENGTH];
          ^
tools/env/fw_env.c:1032:2: error: unknown type name 'uint32_t'
  uint32_t aes_blocks;
  ^
tools/env/fw_env.c:1035:26: error: 'key_exp' undeclared (first use in this function)
  aes_expand_key(aes_key, key_exp);
                          ^
tools/env/fw_env.c:1035:2: warning: passing argument 1 of 'aes_expand_key' from incompatible pointer type [enabled by default]
  aes_expand_key(aes_key, key_exp);
  ^
In file included from tools/env/fw_env.c:38:0:
include/aes.h:43:6: note: expected 'u8 *' but argument is of type 'int *'
 void aes_expand_key(u8 *key, u8 *expkey);
      ^
tools/env/fw_env.c:1041:3: warning: passing argument 2 of 'aes_cbc_encrypt_blocks' from incompatible pointer type [enabled by default]
   aes_cbc_encrypt_blocks(key_exp, data, data, aes_blocks);
   ^
In file included from tools/env/fw_env.c:38:0:
include/aes.h:82:6: note: expected 'u8 *' but argument is of type 'int *'
 void aes_cbc_encrypt_blocks(u8 *key_exp, u8 *src, u8 *dst, u32 num_aes_blocks);
      ^
tools/env/fw_env.c:1041:3: warning: passing argument 3 of 'aes_cbc_encrypt_blocks' from incompatible pointer type [enabled by default]
   aes_cbc_encrypt_blocks(key_exp, data, data, aes_blocks);
   ^
In file included from tools/env/fw_env.c:38:0:
include/aes.h:82:6: note: expected 'u8 *' but argument is of type 'int *'
 void aes_cbc_encrypt_blocks(u8 *key_exp, u8 *src, u8 *dst, u32 num_aes_blocks);
      ^
tools/env/fw_env.c:1043:3: warning: passing argument 2 of 'aes_cbc_decrypt_blocks' from incompatible pointer type [enabled by default]
   aes_cbc_decrypt_blocks(key_exp, data, data, aes_blocks);
   ^
In file included from tools/env/fw_env.c:38:0:
include/aes.h:92:6: note: expected 'u8 *' but argument is of type 'int *'
 void aes_cbc_decrypt_blocks(u8 *key_exp, u8 *src, u8 *dst, u32 num_aes_blocks);
      ^
tools/env/fw_env.c:1043:3: warning: passing argument 3 of 'aes_cbc_decrypt_blocks' from incompatible pointer type [enabled by default]
   aes_cbc_decrypt_blocks(key_exp, data, data, aes_blocks);
   ^
In file included from tools/env/fw_env.c:38:0:
include/aes.h:92:6: note: expected 'u8 *' but argument is of type 'int *'
 void aes_cbc_decrypt_blocks(u8 *key_exp, u8 *src, u8 *dst, u32 num_aes_blocks);
      ^
tools/env/fw_env.c: In function 'fw_env_open':
tools/env/fw_env.c:1279:20: error: 'uint8_t' undeclared (first use in this function)
  crc0 = crc32 (0, (uint8_t *) environment.data, ENV_SIZE);
                    ^
tools/env/fw_env.c:1279:29: error: expected expression before ')' token
  crc0 = crc32 (0, (uint8_t *) environment.data, ENV_SIZE);
                             ^
tools/env/fw_env.c:1279:29: error: too few arguments to function 'crc32'
In file included from tools/env/fw_env.c:36:0:
tools/env/fw_env.h:63:23: note: declared here
 extern unsigned long  crc32  (unsigned long, const unsigned char *, unsigned);
                       ^
tools/env/fw_env.c:1336:30: error: expected expression before ')' token
   crc1 = crc32 (0, (uint8_t *) redundant->data, ENV_SIZE);
                              ^
tools/env/fw_env.c:1336:30: error: too few arguments to function 'crc32'
In file included from tools/env/fw_env.c:36:0:
tools/env/fw_env.h:63:23: note: declared here
 extern unsigned long  crc32  (unsigned long, const unsigned char *, unsigned);
                       ^
scripts/Makefile.host:134: recipe for target 'tools/env/fw_env.o' failed
make[5]: *** [tools/env/fw_env.o] Error 1
Makefile:1189: recipe for target 'env' failed
make[4]: *** [env] Error 2
make[4]: Leaving directory '/root/turris/turris-os/build_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/u-boot-2014.10'
Makefile:114: recipe for target '/root/turris/turris-os/build_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/u-boot-2014.10/.built' failed
make[3]: *** [/root/turris/turris-os/build_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/u-boot-2014.10/.built] Error 2
make[3]: Leaving directory '/root/turris/turris-os/package/boot/uboot-envtools'
package/Makefile:193: recipe for target 'package/boot/uboot-envtools/compile' failed
make[2]: *** [package/boot/uboot-envtools/compile] Error 2
make[2]: Leaving directory '/root/turris/turris-os'
package/Makefile:190: recipe for target '/root/turris/turris-os/staging_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/stamp/.package_compile' failed
make[1]: *** [/root/turris/turris-os/staging_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/root/turris/turris-os'
/root/turris/turris-os/include/toplevel.mk:181: recipe for target 'world' failed
make: *** [world] Error 2

```

miska, 2016-10-17:
Looks like it is missing some include from the look of it. Just to make sure we have same config, I'm using `compile_fw omnia` to start building stuff. But in general, the build system breaks a little randomly depending on host system and currently it is not even ready for gcc6.

achim71, 2016-10-17:
Thank you for the response, i'll try the compile_fw_omnia branch. I'm used to the build env from openwrt and started with an small default config.

achim71, 2016-10-17:
Getting the same error.
This is what i did:

```
git clone -b compile_fw_omnia_image https://github.com/CZ-NIC/turris-os
cd turris-os/
./scripts/feeds update -a
./scripts/feeds install -a
make menuconfig
make -j8
```
I select Target System "Marvel Amada 37x/38x/XP" and Target Profile "Turris Omnia"

It fails on base-files because of missing keys key-build and key-build.pub so i created fake files
```
echo 123 > key-build 
echo 123 > key-build.pub
``` 
Should work till the build env tries to sign anything.
Gcc version is 4.9.2

Error is abit different:
```
make[4]: Entering directory '/dpool/data/turris/branches/compile_fw_omina_image/turris-os/build_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/u-boot-2014.10'
  HOSTCC  scripts/basic/fixdep
  HOSTCC  scripts/kconfig/conf.o
  SHIPPED scripts/kconfig/zconf.tab.c
  SHIPPED scripts/kconfig/zconf.lex.c
  SHIPPED scripts/kconfig/zconf.hash.c
  HOSTCC  scripts/kconfig/zconf.tab.o
In file included from scripts/kconfig/zconf.tab.c:2537:0:
scripts/kconfig/menu.c: In function 'get_symbol_str':
scripts/kconfig/menu.c:590:18: warning: 'jump' may be used uninitialized in this function [-Wmaybe-uninitialized]
     jump->offset = strlen(r->s);
                  ^
scripts/kconfig/menu.c:551:19: note: 'jump' was declared here
  struct jump_key *jump;
                   ^
  HOSTLD  scripts/kconfig/conf
scripts/kconfig/conf --silentoldconfig Kconfig
***
*** Configuration file ".config" not found!
***
*** Please run some configurator (e.g. "make oldconfig" or
*** "make menuconfig" or "make xconfig").
***
/dpool/data/turris/branches/compile_fw_omina_image/turris-os/build_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/u-boot-2014.10/scripts/kconfig/Makefile:36: recipe for target 'silentoldconfig' failed
make[6]: *** [silentoldconfig] Error 1
/dpool/data/turris/branches/compile_fw_omina_image/turris-os/build_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/u-boot-2014.10/Makefile:464: recipe for target 'silentoldconfig' failed
make[5]: *** [silentoldconfig] Error 2
  HOSTCC  tools/env/aes.o
  HOSTCC  tools/env/crc32.o
  HOSTCC  tools/env/ctype.o
  HOSTCC  tools/env/env_attr.o
  HOSTCC  tools/env/env_flags.o
  HOSTCC  tools/env/fw_env.o
tools/env/fw_env.c:56:2: error: unknown type name 'uint8_t'
  uint8_t mtd_type;  /* type of the MTD device */
  ^
tools/env/fw_env.c:81:2: error: unknown type name 'uint32_t'
  uint32_t crc; /* CRC32 over data bytes    */
  ^
tools/env/fw_env.c:86:2: error: unknown type name 'uint32_t'
  uint32_t crc; /* CRC32 over data bytes    */
  ^
tools/env/fw_env.c:99:2: error: unknown type name 'uint32_t'
  uint32_t  *crc;
  ^
tools/env/fw_env.c:111:1: error: unknown type name 'uint8_t'
 static uint8_t aes_key[AES_KEY_LENGTH] = { 0 };
[... 90 lines cut ...]
tools/env/fw_env.c:1336:30: error: expected expression before ')' token
   crc1 = crc32 (0, (uint8_t *) redundant->data, ENV_SIZE);
                              ^
tools/env/fw_env.c:1336:30: error: too few arguments to function 'crc32'
In file included from tools/env/fw_env.c:36:0:
tools/env/fw_env.h:63:23: note: declared here
 extern unsigned long  crc32  (unsigned long, const unsigned char *, unsigned);
                       ^
scripts/Makefile.host:134: recipe for target 'tools/env/fw_env.o' failed
make[5]: *** [tools/env/fw_env.o] Error 1
Makefile:1189: recipe for target 'env' failed
make[4]: *** [env] Error 2
make[4]: Leaving directory '/dpool/data/turris/branches/compile_fw_omina_image/turris-os/build_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/u-boot-2014.10'
Makefile:114: recipe for target '/dpool/data/turris/branches/compile_fw_omina_image/turris-os/build_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/u-boot-2014.10/.built' failed
make[3]: *** [/dpool/data/turris/branches/compile_fw_omina_image/turris-os/build_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/u-boot-2014.10/.built] Error 2
make[3]: Leaving directory '/dpool/data/turris/branches/compile_fw_omina_image/turris-os/package/boot/uboot-envtools'
package/Makefile:193: recipe for target 'package/boot/uboot-envtools/compile' failed
make[2]: *** [package/boot/uboot-envtools/compile] Error 2
make[2]: Leaving directory '/dpool/data/turris/branches/compile_fw_omina_image/turris-os'
package/Makefile:190: recipe for target '/dpool/data/turris/branches/compile_fw_omina_image/turris-os/staging_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/stamp/.package_compile' failed
make[1]: *** [/dpool/data/turris/branches/compile_fw_omina_image/turris-os/staging_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/dpool/data/turris/branches/compile_fw_omina_image/turris-os'
/dpool/data/turris/branches/compile_fw_omina_image/turris-os/include/toplevel.mk:181: recipe for target 'world' failed

```

achim71, 2016-10-17:
It looks like it misses the .config from the kernel package

achim71, 2016-10-17:
Ok it needs this patch https://github.com/openwrt/openwrt/commit/d6c4db5f9bdd9ff502043a8ec6f5c948dd2f9e84

achim71, 2016-10-17:
The mtd package needs this similar patch
```
--- mtd/linksys_bootcount.c.orig	2016-10-17 23:38:08.717902102 +0200
+++ mtd/linksys_bootcount.c	2016-10-17 23:31:32.033902102 +0200
@@ -22,6 +22,7 @@
 #include <stdlib.h>
 #include <stddef.h>
 #include <unistd.h>
+#include <stdint.h>
 #include <fcntl.h>
 #include <sys/mman.h>
 #include <sys/stat.h>

```

achim71, 2016-10-17:
To get the packages signed i backported signify-openbsd and ran

`signify-openbsd -G -s key-build -p key-build.pub -n`

With this the last step (package/index) passed and "make world" finished successfull.

achim71, 2016-10-18:
Is there an way to build an medkit for turris-omnia, I only see code medkits for  TURRISNOR/NAND in the mpc85xx target?

achim71, 2016-10-18:
I uploaded an self build openwrt-mvebu-Turris-Omnia-rootfs.tar.gz from the testing branch via usb by renamed to omnia-medkit-latest.tar.gz. Was able to do the basic configuration and move to luci. I have ssh access but after about one minute the device reboots! Can it be there is some initialisation processess running in the background which does not finish proper and reboots the device? Nothing is shown in the log files.

miska, 2016-10-18:
There is a watchdog running that needs to be taken care of. Do you have `watchdog_adjust` service enabled? I think this one let procd know that is should handle it.

achim71, 2016-10-18:
Thanks for pushing the patch for uboot-envtool and the watchdog tip! As said mtd also needs that header file to build proper.
I did not include the following packages required for the watchdog to work: daemon-watchdog, watchdog-adjust and kmod-wdt-orion. Would be nice to have them preselected when selecting the turris-omnia target. Do you prefere separate issues for mtd and target problems?

closed 2016-10-19 by achim71