CZ-NIC/turris-os #8: Include kernel module az6007

ohrensessel, 2016-10-21:
For using tvheadend with my USB DVB-C card I'm dependent on the kernel module az6007, which is not included and there is no kmod-* package available with this module. Would it be possible to provide such a kmod-* package?

sairon, 2016-10-31:
Hello, I will try to look into it soon. Do you have some other Linux machine, so you could connect the tuner and send me the `dmesg` output after connecting the tuner and the difference between `lsmod` output before and after you plug it in?

BKPepe, 2016-11-03:
**_Ubuntu live_**
**Before:**
```
ubuntu@ubuntu:~$ lsmod
Module                  Size  Used by
drbg                   32768  1
ansi_cprng             16384  0
ctr                    16384  1
ccm                    20480  1
arc4                   16384  2
iwldvm                233472  0
mac80211              737280  1 iwldvm
snd_hda_codec_realtek    81920  1
snd_hda_codec_generic    77824  1 snd_hda_codec_realtek
snd_hda_intel          36864  3
iwlwifi               200704  1 iwldvm
snd_hda_codec         135168  3 snd_hda_codec_realtek,snd_hda_codec_generic,snd_hda_intel
snd_hda_core           73728  4 snd_hda_codec_realtek,snd_hda_codec_generic,snd_hda_codec,snd_hda_intel
coretemp               16384  0
snd_hwdep              16384  1 snd_hda_codec
snd_pcm               106496  3 snd_hda_codec,snd_hda_intel,snd_hda_core
cfg80211              565248  3 iwlwifi,mac80211,iwldvm
input_leds             16384  0
joydev                 20480  0
snd_seq_midi           16384  0
serio_raw              16384  0
snd_seq_midi_event     16384  1 snd_seq_midi
snd_rawmidi            32768  1 snd_seq_midi
lpc_ich                24576  0
snd_seq                69632  2 snd_seq_midi_event,snd_seq_midi
snd_seq_device         16384  3 snd_seq,snd_rawmidi,snd_seq_midi
snd_timer              32768  2 snd_pcm,snd_seq
ite_cir                28672  0
snd                    81920  16 snd_hda_codec_realtek,snd_hwdep,snd_timer,snd_pcm,snd_seq,snd_rawmidi,snd_hda_codec_generic,snd_hda_codec,snd_hda_intel,snd_seq_device
rc_core                28672  1 ite_cir
mac_hid                16384  0
soundcore              16384  1 snd
shpchp                 36864  0
parport_pc             32768  0
ppdev                  20480  0
lp                     20480  0
parport                49152  3 lp,ppdev,parport_pc
autofs4                40960  2
squashfs               49152  1
overlay                49152  1
nls_iso8859_1          16384  1
dm_mirror              24576  0
dm_region_hash         24576  1 dm_mirror
dm_log                 20480  2 dm_region_hash,dm_mirror
uas                    24576  0
usb_storage            69632  2 uas
i915                 1208320  5
psmouse               126976  0
ahci                   36864  0
libahci                32768  1 ahci
i2c_algo_bit           16384  1 i915
drm_kms_helper        139264  1 i915
syscopyarea            16384  1 drm_kms_helper
r8169                  81920  0
sysfillrect            16384  1 drm_kms_helper
sysimgblt              16384  1 drm_kms_helper
mii                    16384  1 r8169
fb_sys_fops            16384  1 drm_kms_helper
drm                   360448  7 i915,drm_kms_helper
fjes                   28672  0
video                  40960  1 i915

```
**_After:_**
```
[... 62 lines cut ...]
sysimgblt              16384  1 drm_kms_helper
mii                    16384  1 r8169
fb_sys_fops            16384  1 drm_kms_helper
drm                   360448  7 i915,drm_kms_helper
fjes                   28672  0
video                  40960  1 i915
```

**_Dmesg:_**
```
[  279.900226] usb 2-2: new high-speed USB device number 3 using ehci-pci
[  280.318849] usb 2-2: New USB device found, idVendor=14f7, idProduct=0003
[  280.318863] usb 2-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  280.318870] usb 2-2: Product: CableStar Combo HD CI
[  280.318880] usb 2-2: Manufacturer: TechniSat Digital S.A.
[  280.318886] usb 2-2: SerialNumber: 0008C9D9032B
[  281.404743] usb read operation failed. (-32)
[  281.405394] usb write operation failed. (-32)
[  281.405787] usb write operation failed. (-32)
[  281.405793] usb 2-2: dvb_usb_v2: found a 'Technisat CableStar Combo HD CI' in cold state
[  281.406071] usb 2-2: Direct firmware load for dvb-usb-terratec-h7-az6007.fw failed with error -2
[  281.406076] usb 2-2: dvb_usb_v2: Did not find the firmware file 'dvb-usb-terratec-h7-az6007.fw'. Please see linux/Documentation/dvb/ for more details on firmware-problems. Status -2
[  281.406086] dvb_usb_az6007: probe of 2-2:1.0 failed with error -2
[  281.406118] usbcore: registered new interface driver dvb_usb_az6007]
```

BKPepe, 2016-11-11:
Finally it was added to test branch. ( https://gitlab.labs.nic.cz/turris/openwrt/merge_requests/8 )
Thanks @sairon

sairon, 2016-11-11:
Indeed @BKPepe, compiled successfully in the nightly build, now it just needs to be staged to the stable build, so I'm closing this issue.

closed 2016-11-11 by sairon