CZ-NIC/turris-os #9: Terrible QoS defaults

apenney, 2016-10-24:
Hi,

As per the discussion at https://discourse.labs.nic.cz/t/wondershaper-srsly-qos-aqm/1095/15 it would be nice if we removed wondershaper in exchange for sqm-scripts-extra/luci-app-sqm.  It would be even better to have Cake compiled in, but just switching the packages and making the QoS tab use the newer stuff would have a drastic impact.

fuller42, 2016-10-24:
thx for taking care of this one!

+1

JimmyS83, 2016-11-06:
Please, yes.. +

danrl, 2016-11-06:
Absolutely +1

kilx, 2016-11-07:
+1 for Cake

anarcat, 2016-11-08:
see also this thread: https://lists.bufferbloat.net/pipermail/cerowrt-devel/2016-November/010896.html

anarcat, 2016-11-23:
looks like this was actually fixed here: https://github.com/CZ-NIC/turris-os/commit/b26a6d2db28b36891649f0022329aaaf9c992f83

@miska should this be closed?

miska, 2016-12-08:
Released

closed 2016-12-08 by miska

ValdikSS, 2016-12-11:
Please update "tc" utility, it's too old and at least can't handle cake. sqm-scripts fail silently.

```
/usr/sbin/tc qdisc add dev eth1 root cake bandwidth 100000kbit diffserv4
Unknown qdisc "cake", hence option "bandwidth" is unparsable
```

quietsche, 2016-12-13:
cake queue discipline not seen, although everything installed and reinstalled. see attached picture from other user working and mine not.
```
# ls -l /var/run/sqm/available_qdiscs
-rw-r--r--    1 root     root             0 Dec 12 07:49 cake
-rw-r--r--    1 root     root             0 Dec 12 07:49 codel
-rw-r--r--    1 root     root             0 Dec 12 07:49 fq_codel
-rw-r--r--    1 root     root             0 Dec 12 07:49 pie
-rw-r--r--    1 root     root             0 Dec 12 07:49 sfq
```
[image]
[image]

ValdikSS, 2016-12-13:
@quietsche make sure you have installed `sqm-scripts-extra`, `kmod-sched-cake`, and run `/usr/lib/sqm/update-available-qdiscs`.

quietsche, 2016-12-13:
Already updated available qdiscs.
```
root@kukuzi:~# opkg list-installed | grep 'kmod-sched-cake\|sqm-scripts'
kmod-sched-cake - 4.4.35+2016-10-02-4f62bd17fa34036cb3c8fd4800a709b2734c3de3-1-34abcd5e548fc8ed5390269f3a31d173-15
sqm-scripts - 1.0.7-1
sqm-scripts-extra - 2016-06-08-1
```

ValdikSS, 2016-12-13:
@quietsche I can only recommend you to add `set -x` on top of `/usr/lib/sqm/update-available-qdiscs` to see what's going on. Also check if you see other files except `fq_codel` in `/var/run/sqm/available_qdiscs`, this could be web interface bug.

quietsche, 2016-12-13:
other files seen, output of [log](https://github.com/quietsche/copyandpaste/blob/master/update-available-qdiscs.log) seems good afais

ValdikSS, 2016-12-13:
@quietsche edit `/etc/sqm/sqm.conf` manually then and run `/etc/init.d/sqm start`.

quietsche, 2016-12-13:
already there =/ seems web/luci problem then?
```
~# cat /etc/sqm/sqm.conf
SQM_LIB_DIR=/usr/lib/sqm
SQM_STATE_DIR=/var/run/sqm
SQM_QDISC_STATE_DIR=${SQM_STATE_DIR}/available_qdiscs
SQM_CHECK_QDISCS="fq_codel efq_codel nfq_codel sfq codel ns2_codel pie sfq cake"
SQM_SYSLOG=1
```

ValdikSS, 2016-12-13:
Oh sorry, I wanted to say `/etc/config/sqm`. Set `qdisc` to the value you want and restart sqm-scripts. But remember cake won't work since `tc` is too old.

Bartvz, 2016-12-13:
@quietsche @ValdikSS you could try updating tc as listed [here.](https://github.com/CZ-NIC/turris-os/issues/33)
Atm I am unable to test it.

ValdikSS, 2016-12-13:
@Bartvz it works. Why the hell tc got updated without version bump!?

sairon, 2016-12-14:
@ValdikSS Sorry for that, fixed in 32123c1477cf915726b55aaa18f807fa53a285c8