CZ-NIC/turris-os #10: luci segfaults on Turris Omnia when viewing the ath10k nic

ddrown, 2016-11-01:
ath10k is on channel 52 (5.260GHz)

Viewing .../cgi-bin/luci/admin/network/wireless/radio0.network1 results in a segfault

Core was generated by `/usr/bin/lua /www/cgi-bin/luci'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0xb6d877fc in ?? () from /usr/lib/libiwinfo.so
(gdb) bt
#0  0xb6d877fc in ?? () from /usr/lib/libiwinfo.so
#1  0xb6db6044 in nla_parse () from /usr/lib/libnl-tiny.so
Backtrace stopped: Cannot access memory at address 0x3000c

I backported the the two commits below to the compile_fw_omnia_image branch and verified they fix the crash.

https://github.com/CZ-NIC/turris-os/commit/b897fed2465af605b3e801b09d1340daf5637a64
"iwinfo: Fix crash when asking for txpowerlist"

https://github.com/CZ-NIC/turris-os/commit/3aa407c737ae657de4adee83eaea97b04cafdf91
"iwinfo: Cleanup the patch"

Before:
root@turris:~# iwinfo phy0 txpowerlist
Segmentation fault

After:
root@turris:~# iwinfo phy0 twpowerlist
   0 dBm (   1 mW)
   1 dBm (   1 mW)
....

Please add these patches to your next Omnia update.  Thank you.

miska, 2016-11-02:
Yep, those commits are in the test branch and will be part of next non-security update.

sairon, 2016-11-02:
Hi, thanks for the investigation. All the patches in the `test` branch should be in the following update for Omnia, so I hope it will resolve the issue for you. The `compile_fw_omnia_image` was some forgotten branch, sorry for the confusion. The production branch for the routers is `stable` (or currently `stable-omnia`).

closed 2016-11-02 by sairon

sjug, 2016-11-21:
When is the next update scheduled?