CZ-NIC/turris-os #11: How to upstream my changes?

danrl, 2016-11-06:
I am the author of [luci-proto-wireguard ](https://github.com/danrl/luci-proto-wireguard) and would like to integrate this functionality into Turris Omnia. What is the best way to go?

Push it to OpenWRT and wait until it gets copied to turris-os? Will it get copied at all?

miska, 2016-11-07:
If it will end up in LuCI or packages repository, it will get merged. From base OpenWRT/LEDE repository it will eventually get merged as well, just later as it is much easier to update feeds.

danrl, 2016-11-07:
Thanks @mista, one file will go into packages, the two other files will go into luci. No need to push something to base. Looking forward to see it on my Omnia eventually 👍

I'll update/close the issue once the files underwent all testing and have been accepted by upstream.

danrl, 2016-11-24:
Files have been accepted by upstream. Affected packages are `wireguard-tools` and `luci-proto-wireguard`.

closed 2016-11-24 by danrl