CZ-NIC/turris-os #14: Kernel Oops

ohrensessel, 2016-11-09:
Hi, my Turris Omnia is seeing Kernel Oops leading to a reboot. So far it seems this only happens when there are clients connected. I will do a factory reset and see whether the problem remains this evening.

```
<1>[ 3794.636389] [6e6573c8] *pgd=00000000
<0>[ 3794.639985] Internal error: Oops: 5 [#1] SMP ARM
<4>[ 3794.644611] Modules linked in: qcserial option iptable_nat ip6table_nat dvb_usb_a800 ath9k w1_gpio usb_wwan usb_serial_simple rndis_host qmi_wwan pppoe nf_nat_pptp nf_nat_ipv6 nf_nat_ipv4 nf_nat_amanda nf_conntrack_pptp nf_conntrack_ipv6 nf_conntrack_ipv4 nf_conntrack_amanda mct_u232 ipt_REJECT ipt_MASQUERADE ebtable_nat ebtable_filter ebtable_broute dvb_usb_v2 dvb_usb_dibusb_mc dvb_usb_dibusb_mb dvb_usb_dibusb_common dvb_usb_dib0700 dvb_usb dib9000 dib8000 dib7000p dib7000m dib3000mc cdc_ether ath9k_common armada_thermal xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_socket xt_recent xt_policy xt_nfacct xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_id xt_hl xt_helper xt_esp xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_comment xt_TPROXY xt_TCPMSS xt_REDIRECT xt_LOG xt_HL xt_DSCP xt_CT xt_CLASSIFY wire usbserial usbnet ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda ts_kmp ts_fsm ts_bm thermal_sys rtl2832 rc_core pppox ppp_mppe ppp_async nfnetlink_acct nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_redirect nf_nat_proto_gre nf_nat_masquerade_ipv4 nf_nat_irc nf_nat_h323 nf_nat_ftp nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_rtcache nf_conntrack_proto_gre nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast mvsdio iptable_raw iptable_mangle iptable_filter ipt_ah ipt_ECN ip_tables hwmon ebtables ebt_vlan ebt_stp ebt_snat ebt_redirect ebt_pkttype ebt_mark_m ebt_mark ebt_limit ebt_ip6 ebt_ip ebt_dnat ebt_arpreply ebt_arp ebt_among ebt_802_3 dvb_core dibx000_common dib3000mb dib0090 dib0070 crc_ccitt cdc_wdm cdc_acm ath9k_hw fuse sch_teql sch_tbf sch_red sch_prio sch_pie sch_netem sch_htb sch_gred sch_fq sch_dsmark sch_codel em_text em_nbyte em_meta em_cmp cls_basic act_ipt act_skbedit act_mirred em_u32 cls_u32 cls_tcindex cls_flow cls_route cls_fw sch_hfsc sch_ingress ath10k_pci ath10k_core mac80211 ath ledtrig_usbdev ledtrig_oneshot xt_LED ledtrig_morse ledtrig_heartbeat ledtrig_gpio cfg80211 compat xt_set ip_set_list_set ip_set_hash_netiface ip_set_hash_netport ip_set_hash_netnet ip_set_hash_net ip_set_hash_netportnet ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set ip6t_NPT ip6t_MASQUERADE nf_nat_masquerade_ipv6 nf_nat nf_conntrack ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_raw ip6table_mangle ip6table_filter ip6_tables pppoatm ppp_generic slhc nfsd nfsv3 nfs msdos ip_gre gre ifb sit ipcomp6 xfrm6_tunnel xfrm6_mode_tunnel xfrm6_mode_transport xfrm6_mode_beet esp6 ah6 ipcomp xfrm4_tunnel xfrm4_mode_tunnel xfrm4_mode_transport xfrm4_mode_beet esp4 ah4 ip6_tunnel tunnel6 tunnel4 ip_tunnel veth tun snd_compress snd_pcm_oss snd_mixer_oss snd_pcm snd_timer snd_rawmidi snd_seq_device snd_hwdep snd input_core soundcore rxkad af_key xfrm_user xfrm_ipcomp xfrm_algo vfat fat udf crc_itu_t ntfs lockd sunrpc grace minix isofs hfsplus hfs cramfs configfs cifs autofs4 kafs af_rxrpc dns_resolver dm_crypt dm_mirror dm_region_hash dm_log dm_mod br2684 atm multipath fscache raid456 async_raid6_recov async_pq async_xor async_memcpy async_tx raid10 raid1 raid0 linear md_mod xt_NFLOG x_tables nfnetlink_log nfnetlink nls_utf8 nls_koi8_r nls_cp1255 nls_iso8859_6 nls_iso8859_2 nls_iso8859_15 nls_iso8859_13 nls_iso8859_1 nls_cp932 nls_cp866 nls_cp864 nls_cp862 nls_cp852 nls_cp850 nls_cp775 nls_cp437 nls_cp1251 nls_cp1250 jitterentropy_rng xts algif_skcipher algif_hash af_alg sha512_generic sha256_generic sha1_generic seqiv pcbc cryptosoft cryptodev ocf md5 md4 marvell_cesa hmac gf128mul fcrypt ecb des_generic ctr cmac ccm cbc authenc usb_storage xhci_plat_hcd xhci_pci xhci_hcd orion_wdt uhci_hcd ledtrig_transient ahci_mvebu ahci ehci_orion ehci_platform ehci_hcd sd_mod ahci_platform libahci_platform libahci libata scsi_mod xfs libcrc32c jfs f2fs usbcore nls_base usb_common mii aead crypto_null [last unloaded: act_police]
<4>[ 3795.009015] CPU: 1 PID: 32241 Comm: curl Not tainted 4.4.13-05df79f63527051ea0071350f86faf76-9 #1
<4>[ 3795.017908] Hardware name: Marvell Armada 380/385 (Device Tree)
<4>[ 3795.023840] task: eda3b200 ti: ea366000 task.ti: ea366000
<4>[ 3795.029256] PC is at crypto_exit_ops+0x10/0x60
<4>[ 3795.033709] LR is at crypto_destroy_tfm+0x44/0x58
<4>[ 3795.038423] pc : [<c02ae658>]    lr : [<c02aea54>]    psr: 20000013
<4>[ 3795.038423] sp : ea367cf8  ip : ea367d08  fp : ea367d04
<4>[ 3795.049926] r10: 00000004  r9 : ed896800  r8 : ea366000
<4>[ 3795.055160] r7 : 00000000  r6 : eb027a78  r5 : 6e657320  r4 : eb027a78
<4>[ 3795.061701] r3 : 65687420  r2 : 6e657320  r1 : eb027a78  r0 : eb027a78
<4>[ 3795.068242] Flags: nzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment none
<4>[ 3795.075392] Control: 10c5387d  Table: 2c58804a  DAC: 00000051
<0>[ 3795.081150] Process curl (pid: 32241, stack limit = 0xea366210)
<0>[ 3795.087081] Stack: (0xea367cf8 to 0xea368000)
<0>[ 3795.091447] 7ce0:                                                       ea367d24 ea367d08
<0>[ 3795.099644] 7d00: c02aea54 c02ae654 00000000 c0039f38 00000010 bf24ae40 ea367d4c ea367d28
<0>[ 3795.107841] 7d20: bf249668 c02aea1c 06000000 bf23d980 ed896800 60000013 06000000 00000000
<0>[ 3795.116038] 7d40: ea367d7c ea367d50 bf23a6b4 bf2494d8 00000000 e968c500 e968c500 bef929bc
<0>[ 3795.124235] 7d60: df237cc0 ec16d340 bef929bc 00000000 ea367d94 ea367d80 bf2420a8 bf23a59c
<0>[ 3795.132431] 7d80: bf244d40 e968c500 ea367f04 ea367d98 bf243224 bf242078 e97c47b0 7ed9859f
<0>[ 3795.140628] 7da0: ea367dd4 effcf960 00000004 06000000 ea367dd4 ea367dc4 00000004 06000000
<0>[ 3795.148824] 7dc0: e420e07c 00080008 00000008 00000000 c00a362c c00a29e8 00000000 00000000
<0>[ 3795.157021] 7de0: 00000000 b6e10000 ee5fac90 00000000 00000000 00000000 00000000 00000000
<0>[ 3795.165218] 7e00: 00000000 00000000 00000000 00000000 00000001 00000040 00000000 df22fcc0
<0>[ 3795.173415] 7e20: 00000000 00000000 00000000 00000000 00000000 c00a344c 00000000 00000000
<0>[ 3795.181612] 7e40: 00080001 00000002 00000003 fffffdfa ef30d9f8 e420e000 00000001 00000000
<0>[ 3795.189808] 7e60: 00000008 b6ebe568 00000000 00000000 00000001 00000000 00000000 00000000
<0>[ 3795.198005] 7e80: 00000000 00000000 00000079 e420e040 00000000 ea367fb0 eda3b200 e9660580
<0>[ 3795.206202] 7ea0: 80000007 b6e17184 e96605b8 e97c47b0 ea367efc ea367ec0 c0020fa0 c005d154
<0>[ 3795.214399] 7ec0: e95e7015 00000000 00000000 00000054 ee0d4af8 00000007 c0020db4 ec16d340
<0>[ 3795.222595] 7ee0: ec16d340 bef929bc 40046366 ee0d4af8 bef929bc 00000000 ea367f14 ea367f08
<0>[ 3795.230792] 7f00: bf2443ac bf24231c ea367f7c ea367f18 c00f43dc bf2443a8 e95e7000 00000003
<0>[ 3795.238989] 7f20: ea367f4c ea367f30 c00ff5e8 c0012fc4 ec16d340 ec16d340 00000000 00000001
<0>[ 3795.247185] 7f40: ea367f7c ea367f50 c00f2a18 c00ff5c4 c00e1c8c ec16d340 ec16d340 bef929bc
<0>[ 3795.255382] 7f60: 40046366 00000004 ea366000 00000000 ea367fa4 ea367f80 c00f445c c00f3d3c
<0>[ 3795.263579] 7f80: bef92990 00000001 0000001f b6ebe460 00000036 c0009ea4 00000000 ea367fa8
<0>[ 3795.271775] 7fa0: c0009ce0 c00f442c 00000001 0000001f 00000004 40046366 bef929bc bef92990
<0>[ 3795.279972] 7fc0: 00000001 0000001f b6ebe460 00000036 00000001 00000004 bef929bc bef929a4
<0>[ 3795.288168] 7fe0: b6ed3a84 bef92978 b6e2c198 b6faf8c8 00000010 00000004 00000000 00000000
<4>[ 3795.296361] Backtrace:
<4>[ 3795.298824] [<c02ae648>] (crypto_exit_ops) from [<c02aea54>] (crypto_destroy_tfm+0x44/0x58)
<4>[ 3795.307255] [<c02aea10>] (crypto_destroy_tfm) from [<bf249668>] (swcr_freesession+0x19c/0x1f0 [cryptosoft])
<4>[ 3795.317017]  r6:bf24ae40 r5:00000010 r4:c0039f38 r3:00000000
<4>[ 3795.322768] [<bf2494cc>] (swcr_freesession [cryptosoft]) from [<bf23a6b4>] (crypto_freesession+0x124/0x1d4 [ocf])
<4>[ 3795.333051]  r8:00000000 r7:06000000 r6:60000013 r5:ed896800 r4:bf23d980 r3:06000000
<4>[ 3795.340891] [<bf23a590>] (crypto_freesession [ocf]) from [<bf2420a8>] (0xbf2420a8)
[... cut ...]
<4>[ 3933.518953]  r10:00000000 r9:dd1d4000 r8:00000000 r7:0000000c r6:b6f077d0 r5:e84c5080
<4>[ 3933.526848]  r4:00000000
<4>[ 3933.529397] [<c03eef7c>] (sock_sendmsg) from [<c03f00f4>] (SyS_sendto+0xbc/0xe0)
<4>[ 3933.536811] [<c03f0038>] (SyS_sendto) from [<c0009ce0>] (ret_fast_syscall+0x0/0x3c)
<4>[ 3933.544482]  r8:c0009ea4 r7:00000122 r6:00000fd4 r5:0000000c r4:b6f077d0
<0>[ 3933.551250] Code: 2a00000b e5911018 e151000e 0a000003 (e5113018)
<4>[ 3933.557507] ---[ end trace 6622c3b5c8cb0f9d ]---```

salekseev, 2016-11-15:
```

ja-pa, 2017-11-06:
After first report we updated kernel and drivers few times and nobody reported this issue, perhaps that solved the problem.

Thanks for reporting.

closed 2017-11-06 by ja-pa