CZ-NIC/turris-os #15: XFS filesystems ignored in LuCI Admin

tomkv, 2016-11-13:
The control panel in System | Mount Points ignores XFS formatted volumes and doesn’t allow to choose them for mounting.

Steps to reproduce:

1. have an XFS-formatted volume available (tested with mSATA attached SSD, partitioned using GPT scheme and the first partition formatted using mkfs.xfs, recognized by blkid),
2. Go to System | Mount Points, scroll to Mount Points section, press the Add button,
3. the UUID, Label and Device combo boxes won’t list the volume.

After reformatting the volume to ext4, everything works as it should.

m-reiter, 2018-02-19:
I have also observed that XFS volumes defined in /etc/config/fstab are not auto mounted at system start. Thought I'd mention it here as it might be related.

ja-pa, 2018-11-09:
Moved to https://gitlab.labs.nic.cz/turris/openwrt/issues/108

closed 2018-11-09 by ja-pa