CZ-NIC/turris-os #16: The rrdtool1 graphs for interfaces do not show interface names

tomkv, 2016-11-13:
The LuCI Statistics | Graphs | Interfaces graphs does not show interface names

[image]

The configuration for generating the above output is following:

```
<Plugin interface>
	IgnoreSelected false
	Interface eth1
	Interface "6in4-wan6"
	Interface ipsec0
</Plugin>

```
From the output, the user cannot distinguish, which graph is for which interface.

wwwdata, 2016-11-23:
Did you find a fix for that @tomkv ?

tomkv, 2016-11-23:
No, I'm just guessing which interface is which, based on the traffic. The graph generators are ELF binaries, not scripts.

JamesT42, 2016-11-27:
Is ist maybe related to my issue that the collectd version is 4, while OpenWRT Chaos Calmer normally has version 5 of collectd?

ja-pa, 2018-11-09:
Moved to https://gitlab.labs.nic.cz/turris/openwrt/issues/107

closed 2018-11-09 by ja-pa