CZ-NIC/turris-os #20: Omnia goes into CPU spinlock during updater.sh run

salekseev, 2016-11-13:
This results in 100% CPU usage and inability to kill -9 curl, also wpa-supplicant stops working at the same time.

```
[40316.784017] curl            R running      0  7071      1 0x00000003
[40316.790404] Backtrace:
[40316.792871] [<c001c130>] (dump_backtrace) from [<c001c328>] (show_stack+0x18/0x1c)
[40316.800456]  r6:c06812c0 r5:c0681400 r4:ef228d80 r3:00000000
[40316.806172] [<c001c310>] (show_stack) from [<c004b340>] (sched_show_task+0xcc/0x100)
[40316.813935] [<c004b274>] (sched_show_task) from [<c004d36c>] (dump_cpu_task+0x40/0x44)
[40316.821867]  r5:c0681400 r4:00000000
[40316.825472] [<c004d32c>] (dump_cpu_task) from [<c006ce50>] (rcu_dump_cpu_stacks+0x70/0xb4)
[40316.833753]  r4:00000000 r3:00000001
[40316.837357] [<c006cde0>] (rcu_dump_cpu_stacks) from [<c0070378>] (rcu_check_callbacks+0x2b0/0x7b4)
[40316.846333]  r9:c06812c0 r8:ef6d4140 r7:c06812c0 r6:2f05f000 r5:c0678e20 r4:c0675140
[40316.854148] [<c00700c8>] (rcu_check_callbacks) from [<c0072b2c>] (update_process_times+0x40/0x6c)
[40316.863038]  r10:c06a97e0 r9:c06a97e0 r8:c06a9760 r7:c2a65db8 r6:ef6d1c80 r5:00000000
[40316.870934]  r4:ef228d80
[40316.873484] [<c0072aec>] (update_process_times) from [<c0081894>] (tick_sched_timer+0x224/0x268)
[40316.882287]  r5:000024ab r4:a074912b
[40316.885891] [<c0081670>] (tick_sched_timer) from [<c0073568>] (__hrtimer_run_queues+0x118/0x1f0)
[40316.894694]  r10:c067852c r9:ef6d1b0c r8:ef6d1b38 r7:ef6d1c80 r6:ef6d1ac0 r5:ef6d1b00
[40316.902590]  r4:ef6d1b0c
[40316.905139] [<c0073450>] (__hrtimer_run_queues) from [<c0073b70>] (hrtimer_interrupt+0xbc/0x20c)
[40316.913942]  r10:ffffffff r9:00000003 r8:ef6d1b38 r7:ef6d1b58 r6:ef6d1ad4 r5:c0672ac0
[40316.921838]  r4:ef6d1ac0
[40316.924387] [<c0073ab4>] (hrtimer_interrupt) from [<c001f2e8>] (twd_handler+0x38/0x48)
[40316.932319]  r10:00000000 r9:f0803100 r8:ef6d6740 r7:00000011 r6:ef008700 r5:c06885e0
[40316.940215]  r4:ef003d00
[40316.942766] [<c001f2b0>] (twd_handler) from [<c0067e98>] (handle_percpu_devid_irq+0x74/0x8c)
[40316.951224] [<c0067e24>] (handle_percpu_devid_irq) from [<c0063c58>] (generic_handle_irq+0x2c/0x3c)
[40316.960288]  r8:ef005000 r7:00000001 r6:00000000 r5:00000000 r4:c067303c r3:c0067e24
[40316.968100] [<c0063c2c>] (generic_handle_irq) from [<c0063f70>] (__handle_domain_irq+0xa0/0xb4)
[40316.976819] [<c0063ed0>] (__handle_domain_irq) from [<c000952c>] (gic_handle_irq+0x5c/0x9c)
[40316.985187]  r8:c2a65db8 r7:f080210c r6:c067902c r5:c068865c r4:f0802100 r3:c2a65db8
[40316.992999] [<c00094d0>] (gic_handle_irq) from [<c000a654>] (__irq_svc+0x54/0x70)
[40317.000497] Exception stack(0xc2a65db8 to 0xc2a65e00)
[40317.005559] 5da0:                                                       e7b784c4 00000000
[40317.013756] 5dc0: 00000000 0000003f e7b784c0 0000004d ef228d80 e7b784c4 00000002 ed310a00
[40317.021953] 5de0: 00000000 c2a65e14 c2a65e18 c2a65e08 c001134c c0012fec 80000013 ffffffff
[40317.030146]  r9:ed310a00 r8:00000002 r7:c2a65dec r6:ffffffff r5:80000013 r4:c0012fec
[40317.037960] [<c0012fb8>] (_raw_spin_lock) from [<c001134c>] (__mutex_lock_slowpath+0x44/0x15c)
[40317.046592] [<c0011308>] (__mutex_lock_slowpath) from [<c001149c>] (mutex_lock+0x38/0x50)
[40317.054786]  r9:ed310a00 r8:00000002 r7:c2a65ebc r6:00000001 r5:0000004d r4:e7b784c0
[40317.062598] [<c0011464>] (mutex_lock) from [<c00eac6c>] (pipe_write+0x30/0x434)
[40317.069921]  r4:e7b784c0 r3:00000000
[40317.073526] [<c00eac3c>] (pipe_write) from [<c00e3d70>] (do_readv_writev+0x104/0x294)
[40317.081372]  r10:00000000 r9:00000000 r8:00000002 r7:c2a65ebc r6:00000001 r5:ed310a00
[40317.089270]  r4:c00eac3c
[40317.091819] [<c00e3c6c>] (do_readv_writev) from [<c00e3fa0>] (vfs_writev+0x44/0x5c)
[40317.099490]  r9:be90d560 r8:00000002 r7:ed310a00 r6:ed310a00 r5:00000000 r4:be90d560
[40317.107302] [<c00e3f5c>] (vfs_writev) from [<c00e4724>] (SyS_writev+0x44/0x84)
[40317.114538]  r4:00000000
[40317.117086] [<c00e46e0>] (SyS_writev) from [<c0009ce0>] (ret_fast_syscall+0x0/0x3c)
[40317.124758]  r9:c2a64000 r8:c0009ea4 r7:00000092 r6:00000002 r5:0000004d r4:be90d560
```

Cynerd, 2016-12-15:
Could you please also post syslog, so we know how updater caused it?

salekseev, 2016-12-15:
Sorry. I got rid of the router already, as wasn't stable enough for my needs.

closed 2016-12-15 by salekseev