CZ-NIC/turris-os #22: batman_adv broken

TobleMiner, 2016-11-17:
Trying to add a mesh interface for batman results in the following error:

```root@turris:~# echo 'bat0' > /sys/class/net/eth0/batman_adv/mesh_iface 
-bash: echo: write error: Out of memory
```

dmesg:
```[137046.543864] device eth0 left promiscuous mode
[137046.543901] br-lan: port 1(eth0) entered disabled state
[137046.544529] batman_adv: bat0: Adding interface: eth0
[137046.544537] batman_adv: bat0: The MTU of interface eth0 is too small (1500) to handle the transport of batman-adv packets. Packets going over this interface will be fragmented on layer2 which could impact the performance. Setting the MTU to 1532 would solve the problem.
[137046.544550] batman_adv: bat0: Interface activated: eth0
[137046.544987] batman_adv: bat0: Interface deactivated: eth0
[137046.544995] batman_adv: bat0: Removing interface: eth0
[137046.545226] batman_adv: Unable to register the batman interface '': -22
[137046.552747] batman_adv: Unable to register the batman interface '': -22
```

And batctl doesn't work at all:
```root@turris:~# batctl if
Error - batman-adv module has not been loaded
```

A little bit of strace later:
```root@turris:~# strace batctl if
execve("/usr/sbin/batctl", ["batctl", "if"], [/* 14 vars */]) = 0
set_tls(0xb6fccb64, 0, 0xb6fcc918, 0, 0xb6fcca98) = 0
set_tid_address(0xb6fccab4)             = 11733
open("/etc/ld-musl-armhf.path", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/lib/libnl-tiny.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/usr/local/lib/libnl-tiny.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = -1 ENOENT (No such file or directory)
open("/usr/lib/libnl-tiny.so", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fcntl64(3, F_SETFD, FD_CLOEXEC)         = 0
fstat64(3, {st_mode=S_IFREG|0755, st_size=24816, ...}) = 0
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\370\32\0\0004\0\0\0"..., 936) = 936
mmap2(NULL, 90112, PROT_READ|PROT_EXEC, MAP_PRIVATE, 3, 0) = 0xb6f3f000
mmap2(0xb6f54000, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0x5000) = 0xb6f54000
close(3)                                = 0
open("/lib/libgcc_s.so.1", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
fcntl64(3, F_SETFD, FD_CLOEXEC)         = 0
fstat64(3, {st_mode=S_IFREG|0644, st_size=45332, ...}) = 0
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\210<\0\0004\0\0\0"..., 936) = 936
mmap2(NULL, 110592, PROT_READ|PROT_EXEC, MAP_PRIVATE, 3, 0) = 0xb6f24000
mmap2(0xb6f3e000, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED, 3, 0xa000) = 0xb6f3e000
close(3)                                = 0
mprotect(0x2c000, 4096, PROT_READ)      = 0
getuid32()                              = 0
getgid32()                              = 0
stat64("/sys/module/batman_adv/version", 0xbea60cd0) = -1 ENOENT (No such file or directory)
writev(2, [{"", 0}, {"Error - batman-adv module has no"..., 46}], 2Error - batman-adv module has not been loaded
) = 46
exit_group(1)                           = ?
+++ exited with 1 +++
```

It seems that batctl expects `/sys/module/batman_adv/version` to exist. I do see this file on my openwrt routers running older batadv versions. I think this is an issue resulting from  using a new kernel without adopting the batctl package.

Still I'm a bit puzzled by the error message resulting from adding the mesh interface manually to a physical interface. This shouldn't happen. Have you got any idea what might be going wrong?

JamesT42, 2016-12-06:
It works for me
`[   50.145819] batman_adv: bat0: Adding interface: fastd_mesh
[   50.145831] batman_adv: bat0: The MTU of interface fastd_mesh is too small (1312) to handle the transport of batman-adv packets. Packets going over this interface will be fragmented on layer2 which could impact the performance. Setting the MTU to 1532 would solve the problem.
[   50.145836] batman_adv: bat0: Interface activated: fastd_mesh`
And batctl works as well.
Maybe show your network config?

TobleMiner, 2016-12-06:
Oh, wow. That's good to hear. I just wanted to make sure the builds aren't broken. Thanks for the info!

closed 2016-12-06 by TobleMiner