CZ-NIC/turris-os #26: Luci WIFI settings broken

the-s-a-m, 2016-11-24:
Chip: Qualcomm Atheros QCA9880 802.11bgnac (radio0)
Configured with this options in turris interface:
WI-F1 1 Enabled
SSID: Puma_concolor
Hide SSID No
2.4 GHz
802.11n/ac mode disabled or 20mhz tested
channel auto

If I want to edit these options in the luci options (with edit):
http://192.168.1.1/cgi-bin/luci/admin/network/wireless

an 500 - Internal Server Error will be shown

I have the following error in the system.log
2016-11-24T17:18:55+01:00 emerg sfpswitch.py[1441]: DEBUG:src/lib/journal.c:123 (journal_open):Opening journal
2016-11-24T17:18:55+01:00 emerg sfpswitch.py[1441]: line not found
2016-11-24T17:18:55+01:00 debug updater[2316]: src/lib/interpreter.c:104 (interpreter_error_result): 
Stack Traceback
===============
(2) Lua function '?' at line 64 of chunk '"logging"]'
	Local variables:
	 err = string: "[string \"transaction\"]:325: No journal to recover"
	 err2string = Lua function '?' (defined at line 45 of chunk "logging"])
	 msg = string: "\
[string \"transaction\"]:325: No journal to recover"
	 (*temporary) = table: 0x35e070  {msg:
[string "transaction"]:325: No journal to recover}
(3)  C function 'function: 0x35caf0'
(4) field C function 'recover'
(5) Lua global 'recover' at line 325 of chunk '"transaction"]'
	Local variables:
	 run_state = table: 0x1298d0  {initialized:true, init:function: 0x129900, lfile:userdata: 0x178568, release:function: 0x156d20 (more...)}
(6) Lua function '?' at line 386 of chunk '"transaction"]'
	Local variables:
	 (*temporary) = Lua function '?' (defined at line 352 of chunk "transaction"])
2016-11-24T17:18:55+01:00 emerg sfpswitch.py[1441]: DEBUG:src/lib/interpreter.c:104 (interpreter_error_result):
2016-11-24T17:18:55+01:00 emerg sfpswitch.py[1441]: Stack Traceback
2016-11-24T17:18:55+01:00 emerg sfpswitch.py[1441]: ===============
2016-11-24T17:18:55+01:00 emerg sfpswitch.py[1441]: (2) Lua function '?' at line 64 of chunk '"logging"]'
2016-11-24T17:18:55+01:00 emerg sfpswitch.py[1441]: 	Local variables:
2016-11-24T17:18:55+01:00 emerg sfpswitch.py[1441]: 	 err = string: "[string \"transaction\"]:325: No journal to recover"
2016-11-24T17:18:55+01:00 emerg sfpswitch.py[1441]: 	 err2string = Lua function '?' (defined at line 45 of chunk "logging"])
2016-11-24T17:18:55+01:00 emerg sfpswitch.py[1441]: 	 msg = string: "\
2016-11-24T17:18:55+01:00 emerg sfpswitch.py[1441]: [string \"transaction\"]:325: No journal to recover"
2016-11-24T17:18:55+01:00 emerg sfpswitch.py[1441]: 	 (*temporary) = table: 0x35e070  {msg:
2016-11-24T17:18:55+01:00 emerg sfpswitch.py[1441]: [string "transaction"]:325: No journal to recover}
2016-11-24T17:18:55+01:00 emerg sfpswitch.py[1441]: (3)  C function 'function: 0x35caf0'
2016-11-24T17:18:55+01:00 emerg sfpswitch.py[1441]: (4) field C function 'recover'
2016-11-24T17:18:55+01:00 emerg sfpswitch.py[1441]: (5) Lua global 'recover' at line 325 of chunk '"transaction"]'
2016-11-24T17:18:55+01:00 emerg sfpswitch.py[1441]: 	Local variables:
2016-11-24T17:18:55+01:00 emerg sfpswitch.py[1441]: 	 run_state = table: 0x1298d0  {initialized:true, init:function: 0x129900, lfile:userdata: 0x178568, release:function: 0x156d20 (more...)}
2016-11-24T17:18:55+01:00 emerg sfpswitch.py[1441]: (6) Lua function '?' at line 386 of chunk '"transaction"]'
2016-11-24T17:18:55+01:00 emerg sfpswitch.py[1441]: 	Local variables:
2016-11-24T17:18:55+01:00 emerg sfpswitch.py[1441]: 	 (*temporary) = Lua function '?' (defined at line 352 of chunk "transaction"])
2016-11-24T17:18:55+01:00 emerg sfpswitch.py[1441]: 
2016-11-24T17:18:55+01:00 crit updater[2316]: src/opkg-trans/main.c:95 (main): 
[string "transaction"]:325: No journal to recover
2016-11-24T17:18:55+01:00 emerg sfpswitch.py[1441]: [31;1mDIE[0m:src/opkg-trans/main.c:95 (main):
2016-11-24T17:18:55+01:00 emerg sfpswitch.py[1441]: [string "transaction"]:325: No journal to recover
2016-11-24T17:18:55+01:00 emerg sfpswitch.py[1441]: Aborted

BuloZB, 2016-11-24:
same issue for me

the-s-a-m, 2016-11-24:
error from root@turris:/tmp/log/lighttpd# cat error.log
2016-11-24 21:30:44: (mod_cgi.c.607) cgi died, pid: 7596

ddrown, 2016-11-24:
This is probably related: luci segfaults on Turris Omnia when viewing the ath10k nic #10

miska, 2016-12-08:
In general, with diplexers it is not a good idea to set 5GHz WiFi to 2.5GHz. But fix released.

closed 2016-12-08 by miska