CZ-NIC/turris-os #27: OpenWRT packages feed stuck in Oct 2014?

JamesT42, 2016-11-27:
I believe the OpenWRT packages feeds are all used in a very old state according to the feeds.conf.default I found.
For example collectd in turris omnia is 4.10.8 and in OpenWRT Chaos Calmer its 5.4.2, which has cost me hours to find out why the statistics have all these bugs with non working graphs etc.

miska, 2016-11-28:
Yep, it is freezed at some specific revision. For releases we need to freeze some state and try to make as much as possible to compile and important stuff to work. It will be unfrozen after release and latest changes will be pulled in for the next one.

sairon, 2016-11-28:
Just a little clarification - collectd is quite stale because it's forked in our packages repository: https://github.com/CZ-NIC/turris-os-packages/blob/test/utils/collectd/Makefile

Suggestion: we should go through the packages in our repo and consider whether they should be synced with upstream or removed from the fork (so they would be replaced by the upstream version - not completely removed).

JamesT42, 2016-11-29:
Please update collectd, the version currently used is from 2010 I believe. And creates many problems.

amq, 2016-12-16:
Please update adblock and openvpn.

The author of adblock [managed to add a support](https://github.com/openwrt/packages/commit/65bcae09867a735b098dd7375756a56ff54ee72f) for Omnia in under 24 hours upon a request two months ago, and then the Omnia still can't receive it...

Cynerd, 2018-02-26:
Please understand our situation when we are not on bleeding edge. We are not rolling upstream OpenWRT yet and because of that most of the fixes or updates are delayed even if upstream updates those. But overal openvpn is already updated and adblock is better maintained as separate fast rolling package as it need much faster release cycle than we can provide. But no matter that updated version of adblock will be in next minor release (3.10 or maybe even before).

JamesT42, 2018-02-26:
Yes thank you for commenting this !one! year old issue, which has never been resolved. collectd is still version 4.something which is now about 8 years old, and you want to tell me that turrisos is not bleeding edge. yeah right. sums up everything that is wrong with turrisos.

JamesT42, 2018-11-07:
Amateurish. Disable issues on github if no one cares about them. Switched to real OpenWrt now.

closed 2018-11-07 by JamesT42

BKPepe, 2019-03-09:
Hello,

I'm so sorry for the late reply! I understand that you'd like to have updated collectd. It seems to me that the word "Amauterish" in this case is useless because it does not help anyone, and I understand it very significantly.

I understand, there is a demand for updating collectd. Patches and pull requests are always welcome. We tried to update it many times over the year or two, unfortunately, it still no go. It breaks so many things that we would spend a lot of time on one package, while we can address more important (security) issues or develop new features for our users. There are many things, which we're currently doing and some of them have even a higher priority than others. You can read about our work in developer updates. They're released at an irregular basis on our forum. Our work as open-source, which is visible on GitHub/Gitlab and as well in OpenWRT repository. These days mostly in OpenWrt packages. 

When you'd like to update some packages, please tell us in [this forum thread](https://forum.turris.cz/t/requesting-new-packages-new-versions/4187) for Turris OS 3.x. I don't want to tell you that we'll update everything, but I and my colleagues are updating packages based on your requests and/or if there's security bug and/or there is a new feature, which we think our users will appreciate. If you want to have a newer version of some package in Turris OS 4.x, you need to ask upstream or send a pull request.

You said that you're using OpenWRT now. I'll appreciate if you can give us a chance now or later with Turris OS 4.0, which is currently in [alpha2](https://forum.turris.cz/t/turris-os-4-0-alpha2-is-out/9580?u=pepe). Any feedback pushes us forward. Since Turris OS 4.0, our operating system is designed as a patchset for OpenWrt. This means that we're using OpenWrt feeds with our patches and our feed, where are mostly our packages as Updater-ng, Foris, Pakon, Schnapps, Honeypot as a Service, Netmetr and so on.

I was curious, what people see to have OpenWrt 18.06.02 or OpenWrt master running on Turris Omnia, so I'd be able to compare it with current Turris OS 4.0 that's why I flashed snapshot to try the latest OpenWrt master and here is the difference:

- No LuCI
- No BTRFS (using snapshots thus rollbacks is not possible), it uses squashfs and without any intervention, you don't have 8 GB for internal storage.
- Using dnsmasq instead of Knot Resolver (used by Cloudflare), which supports DNSSEC, DNS over TLS and so on.
- Not preinstalled packages for SCP, usbutils and so on.
- No Suricata, NextCloud

When you want to update to a new snapshot, you need to do these two following commands or do sysupgrade.
```
opkg update
opkg upgrade
```
Maybe there is missing also something else, but I don't use the snapshot on a daily bases.

In our [Turris OS](https://doc.turris.cz/doc/en/howto/turris_software), there is e.g.

- [Foris](https://demo.turris.cz)
Cloud backups, Pakon, OpenVPN (a few clicks and it's done), Storage plugin, Netmetr, SSH honeypot
- Updater-ng for automatic updates, which we do regularly
- BTRFS filesystem (Schnapps)
- Preinstalled and used syslog-ng
- Used OpenSSL instead of mbedTLS
- Netboot you can boot Turris MOX from Turris Omnia.
and so on.

When we'd release alpha3, you may use branch HBD which is compiled daily and it's the same as the daily snapshot from OpenWrt, but with the difference that the things which I mentioned are there. 

____

You may say I'm biased, but recently I requested to have commit access to OpenWrt packages repository to prove you and others that we're doing things differently for Turris OS 4.0 and also we will create pull requests to backport a few packages from master to 18.06.

JamesT42, 2019-03-09:
Seems you really want an answer???
So you are so incompetent that you can not update collectd from 10 year old version to the one used in upstream OpenWrt? WHAT? Are you serious?

   

- LuCI is in releases, in snapshots it takes a minute to install
- Who cares about knot DNS, it could not even to local name resolving for years
- Packages can be installed, who cares that they are not clogging the install by default
- If you would have upstreamed some things instead of just leeching off of OpenWrt everyone could use your holy Suricata or NextCloud by now
- Foris was useless when I used it, so now plus there
- Updater broke something every update for me, every single time -> useless, a router has to work reliably!!!
- Preinstalled syslog broke Luci integrated log reading because it was integrated poorly
- You can use OpenSSL in OpenWrt, what is your point???
- I would never buy something from you again, why would I boot another great hardware with crap software from yoou?
The only thing better is btrfs, but you see, the lack of commitment to upstream anything is what makes your project so bad I never will touch it again. And I'm not alone.

Cynerd, 2019-03-09:
Then don't buy it. Thank you for your respond and I am going to tell you directly: fuck off.
I am not sure what are you trying to achieve. If you are not happy with product then return it.. sell it.. get rid of it. We are going to respond to negative feedback but if it is just to bash us with false claims such as: knot not supporting local names for years when feature was added one year after initial release. Also you seem to not understand that you are not the only snowflake under the sun, because Foris is useless to you does not mean that it is useless for everyone. You think that I use it? No. I am angry that we are developing it? No I am not, I am not its target audience. You claim that updater broke something every time just tells me that you were editing stuff that you should have not. It is like being angry that gun when it is pointed the other way can kill you. Yep it can, what a surprise. You feel like true power user, I am sure, but in reality you clearly have to still grasp some basic concepts of software development like: stuff take time, there is nothing like unlimited man power and because it does not serves me does not mean that it is useless. How about next time instead of taking our time with this crap you would take your courage and once at the time tried to update that package so you know what are you talking about.

JamesT42, 2019-03-09:
You have demonstrated your level perfectly: it "fuck off". Thanks!

samm-git, 2019-03-09:
omg, such a pain here. Just a few notes from another turris user (and opensource developer and OpenWRT contributor)

1. Yes, it is frustrating to see many packages outdated or just broken on TurrisOS. E.g. i had to build asterisk + many other things myself. However, as it was mentioned before - nobody been promising that they will be up to date, and syncing all packages with an upstream without regressions is a very hard task. 
1. I still think that Turris is one of the best projects ever and using it both at home and in office. And yes, i have v1 and v2 :) 
1. Why to blame developers if you have a lot of choice? E.g.
   - Use OpenWRT (and your hw will still work great!)
   - Build custom feed (thats what i have done, its easy!)
   - Send a PR to specific package or just provide own static build (did it for some golang staff)
1. You need to understand that Turris OS is not just an OpenWRT fork. It is OpenWRT based project with some own specific goals and some in-house components which may and will conflict in some cases with upstream. 
1. After all - it always was this way, so not sure why you bought this device if it is not something you happy with. Go with some crappy *** vendored box where you will need soldering machine + a lot of luck to modify _anything_ non standard. 

@Cynerd just a suggestion to avoid such flames in the public space. It is not beneficial for the project image at all. And keep making best router ever :)

JamesT42, 2019-03-09:
"so not sure why you bought this device"
I funded it on Indiegogo, and back then it was communicated to be OpenWrt based. That they wont upstream any hardware support was not communicated.

samm-git, 2019-03-10:
1. It is OpenWRT based as it was promised
1. Source is open, as well as hardware and there preventing you to run OpenWRT. 
1. You can also run openwrt (or any other arm-compatible distro) in lxc

JamesT42, 2019-03-10:
Thank you for stating the obvious. Could we please stop the propaganda in this issue vom 2.5 years ago, which by the way was closed for good reasons?

samm-git, 2019-03-10:
You can always unsubscribe :) And yes, i got notifications because of *your* comments, not mine.

BKPepe, 2019-03-10:
@JamesT42 I didn't know, why did you come to the conclusion that we're not contributing to OpenWrt at all. There are many ways, how you can look at it. You can look at advanced search here at Github or can look if you wanted to OpenWrt repositories yourself. Didn't I say that I want to backport some packages there? I thought that I explained almost everything in my large post, but I don't think so that you tried Turris OS 4.0, which I told you that it's based on latest stable release of OpenWrt or master, it's what you can choose, but instead of it you told us false claims, which are no true. It's your opinion at all.

@samm-git: Thank you for your feedback! We appreciated.

~ locking this conversation as it doesn't go anywhere.