CZ-NIC/turris-os #32: luci remote logging settings are ignored and never applied to syslog-ng

arekm, 2016-12-09:
Omnia LUCI has remote logging configuration (system->system->logging) which is stored in UCI database but never stored in appropriate /etc/syslog-ng.d/XYZ.conf file.

waja, 2017-06-28:
This issue is a bit more described at https://forum.turris.cz/t/remote-log-how-to-configure/992/7. It would be very appreciated if it would be possible to implement that.

Anyway ... the only part which needs to be modified is ```destination messages``` in ```/etc/syslog-ng.conf```. Redefining those in ```/etc/syslog-ng.d/``` doesn't fit as the first rule matches.

synaptiko, 2017-12-19:
I think that because of this you cannot see logs for `mwan3` in Luci because it internally uses `logread` command. Details here: https://forum.turris.cz/t/combination-of-syslog-ng-luci-system-logging-configuration-mwan3/5975/1

ja-pa, 2018-11-09:
Moved to https://gitlab.labs.nic.cz/turris/openwrt/issues/91

closed 2018-11-09 by ja-pa