CZ-NIC/turris-os #35: tc ipt action not working

dim-geo, 2016-12-15:
I'm trying to use tc so I can change the DSCP field.
tc is not aware of DSCP target. I tried also xt action but to no avail.

/usr/sbin/tc filter add dev ifb4pppoe-wan protocol ip parent 1:0 prio 24 u32 match ip dport 45322 0xffff flowid 1:13 action ipt -j DSCP --set-dscp 32
 failed to find target DSCP

bad action parsing
parse_action: bad value (5:ipt)!
Illegal "action"
root@turris:~# /usr/sbin/tc filter add dev ifb4pppoe-wan protocol ip parent 1:0 prio 24 u32 match ip dport 45322 0xffff flowid 1:13 action ipt -j dscp --set-dscp 32
 failed to find target dscp

bad action parsing
parse_action: bad value (5:ipt)!
Illegal "action"

guidosarducci, 2017-03-10:
Were you able to learn anything more about this issue?  I found it's also a problem in the latest LEDE release unfortunately.  It seems like many tc filter actions (ipt, xt, pedit) have problems...

dim-geo, 2017-03-15:
Nope, I just tried again both pedit, ipt but to no avail.

kernel modules are there but tc cannot see them.

lsmod | grep act
act_bpf                 2914  0 
act_connmark            1873  0 
act_csum                4485  0 
act_gact                2175  0 
act_ipt                 2914  0 
act_mirred              3030  1 
act_nat                 3202  0 
act_pedit               2922  0 
act_police              3711  0 
act_skbedit             1934  0 
act_vlan                2117  0

rshirani, 2018-01-04:
I am hitting the same issue right now. Did anybody figure out what the issue is?

:~$ sudo tc filter add dev eth0 parent ffff: protocol ip prio 10 u32 match u32 0 0 flowid 1:1 action ipt -j MARK --set-mark 1 action mirred egress redirect dev ifb0
bad action type ipt
Usage: ... gact <ACTION> [RAND] [INDEX]
Where: 	ACTION := reclassify | drop | continue | pass 
	RAND := random <RANDTYPE> <ACTION> <VAL>
	RANDTYPE := netrand | determ
	VAL : = value not exceeding 10000
	INDEX := index value used

bad action parsing
parse_action: bad value (11:ipt)!
Illegal "action"

guidosarducci, 2018-01-21:
I've found DSCP-related tc filter actions (ipt, xt, pedit) work on some distros, not on others, with little consistency. I needed portability so couldn't use those actions in the end. Part of the issue seems that the iproute/tc code tracks the latest kernel, while back-porting to older kernels is very dependent on distro and kernel.  Using a recent kernel/iproute combination would be most likely to have everything working, but that didn't help much when using an OpenWRT-derivative.

ja-pa, 2018-11-09:
Moved to https://gitlab.labs.nic.cz/turris/openwrt/issues/88

closed 2018-11-09 by ja-pa