CZ-NIC/turris-os #40: Confusing build instructions

Shulyaka, 2017-01-11:
Hi,

The README file suggests to use 'make menuconfig && make'. However there is no turris omnia board profile in the menuconfig, and as the result 'make' builds a MIPS image by default.

I understand that the README file was copied from OpenWRT project and was not modified since then.
I also understand that I probably need to run 'compile_omnia_fw' script. I have managed to build the image with that (which also required patching of bundled gcc), however I'm still not sure how to select additional packages, because the script calls 'make defconfig' every time which overwrites any previous configuration.

Also I am not sure how to verify if the build is complete. I can see the image file in bin/mvebu-musl/openwrt-mvebu-turris-omnia-rootfs.ext4 and packages in bin/mvebu-musl/packages/ however it doesn't seem like the 'compile_fw' script has completed successfully (it stops after calling ./scripts/generate_junit.sh).

It would be good if the old OpenWRT build instructions are replaced with at least short message like 'To compile run './compile_fw [turris|omnia]', the built image is located at ***'. It would be even better if turris stuff are integrated into OpenWRT build system (not opposite) eliminating the need of a custom script.

Best regards,
Denis Shulyaka

miska, 2017-01-12:
You still can use menuconfig and make, but you need to select Marvell Armada target system first and only after that you will be able to select profile for Omnia.

`compile_fw` script is what we use and what produces based on some undocumented variables images and packages as we publishe them. Yes, documentation could be better and it would be even better to integrate everything in build system, but regarding the integration we still have more important issues at hand.

ja-pa, 2017-10-13:
There is also howto on our wiki https://www.turris.cz/doc/en/howto/turris_os_build . I' closing this issue, but feel free to create new one if you find other problems.  Patches are also welcome.

closed 2017-10-13 by ja-pa