CZ-NIC/turris-os #41: Enable /dev/disk/by-partuuid/ Support

fmfranz, 2017-01-14:
Please enable support for /dev/disk/by-partuuid/ folder

Cynerd, 2018-04-11:
This is provided by udev. You can install udev on Turris but it's not well supported. There is not even a init script for it. So you have to configure and run it on your own.
If you want better support for udev (or probably better eudev) then contact upstream OpenWRT and check their status. Turris OS 4.0 will be based on upstream OpenWRT and before that there is probably no incentive to fix or improve udev in current stable Turris OS. But of course patches are welcome.

closed 2018-04-11 by VojtechMyslivec

bernd-wechner, 2018-04-11:
Awesome to hear Turris OS is merging in upstream OpenWRT again. As ever am curious if there are timelines for 4.0 and if this will mean we win a solution for:

https://forum.turris.cz/t/etc-init-d-rcs-is-missing-what-gives/4745/11

which is essentially about plugging in a USB drive linking /var to it not /tmp early in the boot sequence before any logs are written (who doesn't want persistent logs?).

Cynerd, 2018-04-11:
No Turris OS is not merging with OpenWRT. It's going to be build from scratch on top of new OpenWRT. Current fork is going to be obsoleted and dropped.
This is somewhat unrelated issue to this one. I really don't know if there is some documented way in OpenWRT. Current state is also no way Turris OS doing. In reality we haven't touched mounting at all. So I haven't investigated it in depth but I suspect that there is no immediate improvements to your problem. But you can look to upstream OpenWRT to see.

bernd-wechner, 2018-04-11:
Don't worry, that's more or less what I understood, albeit a little more radical.

Happy to check on OpenWRT too. Because it's a common router issue, but for Omnia perhaps especially relevant given you have USB ports and support plug in storage. I mean who doens't want log persistence across reboots?

Cynerd, 2018-04-12:
You know that if you want just persistent logs that that is easy to be done? We are using syslog-ng so you can just add another destination to syslog-ng configuration file and that is it. No hustle with mounting /var what so ever.

bernd-wechner, 2018-04-12:
Well that be a super start. Got a pointer to some doco?

It's not all, I would like stuff like the majordomo database on persistent store too, and basically anything else that in under /var as a rule, but yes, can do them one by one configuring each one to be somewhere else, and will do if that's what it takes.  Thanks for the tip though! I'll look at syslog-ng and how to configure that. Persistence here we come ;-).