CZ-NIC/turris-os #48: OpenSSL path Hardcoded?

ZackMattor, 2017-02-02:
When i run make on open-wrt master it configures just fine on my system. But on the turris-os branch here it finds all deps except openssl? Is there a way to configure this path?

Cynerd, 2018-09-27:
Isa it possible that you are compiling it on system wit OpenSSL 1.1? This system requires OpenSSL 1.0 most probably and because of that any package that needs those packages is going to fail.

Please consider following:
https://doc.turris.cz/doc/en/howto/turris_os_build

bernrad, 2018-09-27:
I am out of the office until 30/09/2018.

I will respond to your message when I return.

Note: This is an automated response to your message  "Re:
[CZ-NIC/turris-os] OpenSSL path Hardcoded? (#48)" sent on 27/09/2018
16:31:00.

This is the only notification you will receive while this person is away.

ja-pa, 2018-11-09:
Moved to https://gitlab.labs.nic.cz/turris/openwrt/issues/75

closed 2018-11-09 by ja-pa