CZ-NIC/turris-os #50: No Support for Lede/OpenWRT

fuller42, 2017-02-03:
the chance to get into the first Lede Release has passed and we a aproaching half a year after shipping.
Still there is no recent sign of progress towards giving back to the community a straigtforward way to install the industry-standard basline distribution (that happens to be upstream for most of the product).

As commonly recognized, such (extended) commitment to upstream as much as possible, would also help the turris project in reducing workload on backporting issues and maintaining baseline functionality.

danrl, 2017-02-21:
This is a big issue and i wholeheartedly agree with OP. We should be able to run LEDE on our Turris Omnia routers!

oerdnj, 2017-02-21:
I am working on merging TurrisOS tree on top of LEDE tree. It will take me some time as this is mostly manual work and I also gave plenty of other duties.

fuller42, 2017-02-21:
> I am working on merging TurrisOS tree on top of LEDE tree

that is big news! thank you
> It will take me some time as this is mostly manual work and I also gave plenty of other duties.

but please dont use past mistakes to rationalize present ones ;)

danrl, 2017-02-21:
Awesome! Thank you @oerdnj, I know it is some tedious work as I tried it myself and gave up due to lack of knowledge of some Turris parts.

mirlang, 2017-02-21:
> I am working on merging TurrisOS tree on top of LEDE tree. It will take me some time as this is mostly manual work and I also gave plenty of other duties.

with linux-4.9 and DSA support? would be even greater :)

Tasqa, 2017-02-27:
There is already some work done here: https://forum.lede-project.org/t/need-help-with-upstreaming-turris-omnia/705/14

amq, 2017-03-24:
Seems like a month has passed without an activity, judging from oerdnj's fork, `dev-ondrej` and `test-lede-17.01` branches

danrl, 2017-03-24:
😩😭

sjug, 2017-04-05:
I would really like to see this happen so we don't need to be held hostage on TurrisOS...

danrl, 2017-04-12:
Any news? Is there anything we can test to help you?

keelson, 2017-05-01:
I would also like to know when LEDE will be supported on the omnia,because I'm currently planning to buy a new router. The omnia would be my favorite,not just because of the specs, I would also like to support the idea behind it. But I am planning to run LEDE on it,so currently I am looking for altenatives.

JamesT42, 2017-05-10:
Seems like the devs won't do it, see recent forum posts. Also lack of any response here sends a clear message. They will only continue with their fork.

danrl, 2017-05-10:
Not cool.

sjug, 2017-05-10:
I guess they feel that only way they would have any substantial user base is to hold everyone hostage.

When I have some free time I will have to try and figure out how to build and port to LEDE.

JamesT42, 2017-05-10:
I've created a new topic to bundle efforts:
https://forum.turris.cz/t/integrating-hardware-support-into-lede-by-the-community/4196

amq, 2017-05-11:
The worst thing is that CZ-NIC gave a false impression of working on this. If CZ-NIC clearly stated that this is up to the community, it would have been probably done a long time ago.

sjug, 2017-05-11:
Look at any other **real** open source companies, they merge everything upstream first. That way they can share the common burden and don't have to maintain everything themselves, as it becomes a community effort at that point. Speaking of which, real open source companies also have substantial contributors from outside the organization.

oerdnj, 2017-05-11:
Folks, please stop with personal attacks. It is neither appropriate nor helpful. 

Nobody disagrees that this needs to be done. We know that there's a huge technical debt that needs to be resolved, but it is not that easy. I spent a substantial amount of time trying to merge the trees, but in the end I gave up as I hadn't had enough time to finish this properly. I apologize for give the hopes back then, I should have come forward sooner.

The problem is that the two repositories (TurrisOS and LEDE) diverged and doesn't have common ancestor (BTW neither OpenWRT and LEDE git trees have common ancestor), so TurrisOS needs to be merged file by file, which is huge and tedious work. 

I understand your displeasure, but unfortunately as far as I understand there are more urgent tasks that needs to be fixed first.

I can assure you that we at CZ.NIC haven't forgotten about this issue, but there are just more pressing issues at hand.

JFTR there's upcoming Turris Hackaton (the first one is Czech only) and this is one of the topics I want to propose to work on, and if there's more people who want to work on that we might be able to clear this long standing debt with the OpenWRT/LEDE community.

amq, 2017-05-11:
Maybe we are talking about slightly different things? You mean rebasing turris-os on LEDE with all its features, but what would be enough is upstreaming the device support, so that people can build a plain LEDE which could be flashed on Turris Omnia.

danrl, 2017-05-11:
Device support is all we ask for. No btrfs magic needed, no rainbow support required. We live these features, but running LEDE master would be enough for now. We can port all the other stuff over if and when it is suitable. That's something the community will be much larger for once we have something running to develop on.

May I ask you to point us to the work that has been done so far, so that we can build upon that? I saw something a while ago but my device refused to boot the resulting medkit image.

oerdnj, 2017-05-11:
I am travelling for DNS related stuff up until next week, but here's my WIP tree: https://github.com/oerdnj/lede/tree/lede

I haven't touched it for a while, and unfortunately I started with the old (blue) Turris, so there's nothiing related to Omnia right now. But if you want to help, it might be a good starting point.

Perhaps we can create an online room (gitter?) for this effort I can give some pointers how the people from community can help cherry-picking the changes and right files? This is outside the main Turris team initiative, but would that be acceptable?

danrl, 2017-05-11:
Thank you very much for sharing @oerdnj!

helb, 2017-05-15:
It seems that [OpenWrt and LEDE will remerge](https://lwn.net/Articles/722135/) after all, so it might be a good idea to (re)base these ongoing efforts on their merged tree.

```
*) timeline
- refine / vote / agree on the proposal withing the next 2 week
- work on the action items in the 4 weeks after that
```
*(posted on May 8)*

rotanid, 2017-05-16:
@helb there won't be a merge regarding the tree. the tree will be the one LEDE is using at the moment.

localnet, 2017-07-10:
@oerdnj, can you collaborate to get SFP working? I am not using it, but I like have support for it.

Here a repo with LEDE for Turris Omnia, working:
https://github.com/rmounce/lede-source

Here the built medkit, working:
https://github.com/rmounce/lede-source/releases/tag/v0.1

Here the PR to official LEDE repo:
https://github.com/lede-project/source/pull/1210

Here the forum post:
https://forum.turris.cz/t/integrating-hardware-support-into-lede-openwrt-by-the-community/4196

rmounce, 2017-07-11:
> can you collaborate to get SFP working on LEDE? I am not using it. But I like have support for it, in a future... who knows?

See some information from a CZ.NIC developer here: https://www.spinics.net/lists/arm-kernel/msg542466.html

> We have our proprietary support hacked onto mvneta driver for disconnecting PHY on the fly. It is a bit nasty, so I suggest to ignore SFP in this DTS altogether and let's wait till "phylink based SFP module support" or something alike hits upstream, so we can base the SFP support on solid code; unless somebody has a better idea, of course.

phylink is not yet in the mainline kernel as I understand, however it has been added in LEDE to support the ClearFog Pro (which is based on very similar hardware to the Omnia). The ClearFog's SFP port is not multiplexed with a copper PHY as the Omnia, so some further patching would be necessary. If a later release of Turris OS was re-based on LEDE and took the approach of patching phylink rather than patching the mvneta driver, then this would be trivial to move to upstream LEDE and in a better position if phylink eventually makes it into the mainline kernel.

localnet, 2017-07-12:
Ok @rmounce, now I understand the situation. Thanks by aclaration.

APTX, 2018-03-05:
Current pull request for OpenWRT:
https://github.com/openwrt/openwrt/pull/749

ja-pa, 2018-03-19:
Basic support of Turris Omnia was merged into Openwrt see https://github.com/openwrt/openwrt/commit/9f3f61a0d968fbe7b93899f948f3c34612683ba6

closed 2018-03-19 by ja-pa