CZ-NIC/turris-os #54: Broken "Drop invalid packets" switch, NAT leakage

ValdikSS, 2017-04-11:
The firewall option "Drop invalid packets" doesn't do anything. It doesn't add rule to drop INVALID packets to the firewall.
I've found this because I observed NAT leakage, the packets with internal IP behind NAT were coming out to the WAN interface.
https://github.com/openwrt/luci/issues/1068

chrootLogin, 2018-06-09:
This is important!

Cynerd, 2018-09-27:
This is Luci related problem and fix is in incompatible Lede 17.01 version. Because of that it won't fix.

Fix is going to be part of 4.0.

Dropping invalid packets is really good idea but if given rule added in Luci is not propagated it might not be as bad as if for example default rule would not be propagated. It is important but to pull changes from 17.01 we would have to update a lot of stuff that would for sure break some other things. Turris 4.0 is a way to go instead.

closed 2018-09-27 by VojtechMyslivec

bernrad, 2018-09-27:
I am out of the office until 30/09/2018.

I will respond to your message when I return.

Note: This is an automated response to your message  "Re:
[CZ-NIC/turris-os] Broken "Drop invalid packets" switch, NAT leakage (#54)"
sent on 27/09/2018 17:28:46.

This is the only notification you will receive while this person is away.

bernrad, 2018-09-27:
I am out of the office until 30/09/2018.

I will respond to your message when I return.

Note: This is an automated response to your message  "Re:
[CZ-NIC/turris-os] Broken "Drop invalid packets" switch, NAT leakage (#54)"
sent on 27/09/2018 17:27:47.

This is the only notification you will receive while this person is away.