CZ-NIC/turris-os #55: Update Wireguard

fmfranz, 2017-04-16:
The included version doesn't work with new versions anymore

anarcat, 2017-10-24:
this was also discussed in the following threads:

* https://forum.turris.cz/t/update-wireguard-package/3606/12
* https://forum.turris.cz/t/experiences-with-wireguard/3318/6
* https://forum.turris.cz/t/requesting-new-packages-new-versions/4187/19
* https://gitlab.labs.nic.cz/turris/turris-os-packages/issues/44

it also seems we miss the PDATA kernel config option, as documented in https://github.com/CZ-NIC/turris-os/issues/53.

there's also the issue of the missing luci package, documented here:

* https://github.com/CZ-NIC/turris-os/issues/11
* https://forum.turris.cz/t/wireguard-vpn-integration-in-luci/1251

i can confirm that the older version (0.0.20160630-1) doesn't work with newer wireguard releases. after talking briefly with upstream about this, it seems that snapshots from here on (e.g. 201710XX) should be more stable protocol-wise.

zx2c4, 2017-10-24:
Hey Turris,

WireGuard upstream author here.

I wasn't aware that you were shipping the _first ever WireGuard snapshot_, whose git tag explicitly marked it as "experimental". Yowza.

Fortunately, things have calmed down, and the latest releases, which are now part of LEDE's core repo, are considerably more stable. I'd highly recommend you don't delay updating.

Thanks,
Jason

closed 2017-10-24 by CZNIC-GitLab

ja-pa, 2017-10-25:
Thanks for information. I just merged version 0.0.20171017 to our test branch and it should be part of next release.

commit c04d61a0 (2017-10-24): wireguard: update to version from LEDE
merge wireguard package from LEDE

wireguard is now in the base LEDE system

this is a port of https://github.com/lede-project/source/pull/1409

fixes #55
commit e53305c0 (2017-10-24): wireguard: update to version from LEDE
merge wireguard package from LEDE

wireguard is now in the base LEDE system

this is a port of https://github.com/lede-project/source/pull/1409

fixes #55