CZ-NIC/turris-os #63: RAID disks do not spin down

stefan-sedlak, 2017-10-17:
The issue is based on thread:
https://forum.turris.cz/t/raid-disks-do-not-spin-down/5325

After upgrade to 3.8.1/3.8.2 I noticed, that my mdraid disks never spins down. I did some research, but I'm not satisfied with it. Every 10 minutes kernel writes some sector on /dev/md0 and I do not know why.
echo 1 > /proc/sys/vm/block_dump

to debug block activity on disks, and what I found is:

```
2017-10-11T22:10:04+02:00 debug kernel[]: [14952.155530] md0_raid1(2126): WRITE block 8 on sda (1 sectors)
2017-10-11T22:10:04+02:00 debug kernel[]: [14952.161324] md0_raid1(2126): WRITE block 8 on sdb (1 sectors)
2017-10-11T22:10:05+02:00 debug kernel[]: [14953.242609] md0_raid1(2126): WRITE block 8 on sda (1 sectors)
2017-10-11T22:10:05+02:00 debug kernel[]: [14953.248419] md0_raid1(2126): WRITE block 8 on sdb (1 sectors)

```
And this happens every 10 minutes...

Of course, my RAID1 md0 consists of /dev/sda and /dev/sdb. Raid is in "clean" state and is under regular checking. Filesystem is ext4.

Is it some bug of new kernel (kernel module) or am I missing something?

Additional research results:
I just manually span down disks to obtain some seconds to catch process with
```
watch -n1 -d "ps axu | awk '{if (\$8==\"D\") {print \$0}}'"
```
and result is similar each time:
```
USER PID %CPU %MEM VSZ RSS TTY STAT START TIME COMMAND
root 2102 0.0 0.0 0 0 ? D 10:09 0:00 [md0_raid1]
root 27829 0.0 0.0 0 0 ? D 12:55 0:00 [kworker/0:0]
root 29695 0.0 0.0 0 0 ? D 13:02 0:00 [kworker/0:2]
root 31535 0.0 0.0 908 488 ? D 13:05 0:00 /sbin/uci commit dhcp
```

It seems that `/sbin/uci commit dhcp` has something to do with that problem, because it has start time (13:05) whenever a disks spins up.

It is annoying and sector 8 on both disks can be damaged.

ja-pa, 2017-10-21:
hmm that's really strange. I think there wasn't any change in 3.8.1 and 3.8.2 which should affect this.

There is similar topic in czech version of forum https://forum.turris.cz/t/ne-uspavani-disku-hd-idle/3533/12

It'll take some time to reproduce this, because I don't have any hdd nearby.. (maybe @Cynerd can help ?)

You can check  that APM is set to 127 (https://wiki.archlinux.org/index.php/Hdparm)

and try if hd-idle configuration can affect that behavior https://wiki.openwrt.org/doc/uci/hd-idle

stefan-sedlak, 2017-10-25:
I have downgraded Turris Omnia to 3.8.1 and 3.8.2
The issue is reproducible only on 3.8.2

I tried both hdparm and hd-idle before filling the issue. It did not help.
I was following archlinux doc :) I use manjaro at desktop.
Once a process wants to write to disk kernel must wake up disks.

offtopic: doc deserves to mention that if you want to idle muiltiple hard drives you must remove option 'disk'
Next week I am on vacation, so I could spend a time with testing if it helped you.

ja-pa, 2017-10-29:
Right now our main suspect is watchdog, it  sometimes periodicaly restarts kresd. Could you please replace in file **/etc/init.d/resolver**

`pid_num=$(pidof kresd )` with `pid_num=$(pidof kresd |awk '{print $1}' )`

https://github.com/CZ-NIC/turris-os-packages/blob/test/net/resolver-conf/files/resolver-init#L112

stefan-sedlak, 2017-10-29:
I am amazed/surprised. It helped.
How is kresd related to md0?

I recorded htop on 3.8.1 via:
`asciinema rec -c 'ssh root@192.168.1.1 -t htop -d 1' htop.json`
just unzip [htop.json.gz](https://github.com/CZ-NIC/turris-os/files/1425330/htop.json.gz) and play:
`asciinema play htop.json`
There was long delay on /sbin/uci commit dhcp - it confused me.

Thanks!
Now I can have updated OS (3.8.4) without anoying spinning HDDs

ja-pa, 2017-11-02:
Turris OS 3.8.5 with fix for this issue is now in RC

closed 2017-11-06 by ja-pa