CZ-NIC/turris-os #66: How to specify list of packages to be included?

jmguzmanc, 2017-10-30:
Before ad88b93e062ad1312567e0da0808b4884c26799d (@Cynerd), I could specify the list of packages I would like to include in Omnia built. For that compile_fw would create the proper CONFIG_PACKAGE_$PKG settings in .config.

In ad88b93e062ad1312567e0da0808b4884c26799d that was removed.

Is there any simpler (and standardized) way to achieve the same thing now?

Cynerd, 2017-10-30:
Your referred commit has nothing to do with ability to specify list of packages to be installed. In fact it does complete opposite. It enables you to do the same for both small and complete builds. Commit you are looking for is this: 6c74cd63b521c099ea8f3ac8f3533e339ae4adbe

We had to rework how lists were generated as previous implementation was problematic. Currently we have it in lua syntax. But that is no way mirrored to `.config`. These parts are now separate (might join in future but not at the moment). So correct way is adding `CONFIG_PACKAGE_mypkg=y` to `configs/common`.  It's basically the same thing as the original script did so there is no degradation. But in both cases it's just hack as this no way checks dependencies and if they are invalid then kconfig just bails out.

jmguzmanc, 2017-10-30:
My bad, yes, I was referring to  6c74cd63b521c099ea8f3ac8f3533e339ae4adbe !!

I create my configs to configs/common, and worked perfectly.

Thanks a lot!

closed 2017-10-30 by jmguzmanc