CZ-NIC/turris-os #67: Openvswitch not working

jmguzmanc, 2017-10-30:
Hi Guys

I have been stuck on this for days...
I compiled Omnia with openvswitch 2.5.0 userspace tools. (stable-3.8.5)
Everything get compiled and installed

```
kmod-openvswitch - 4.4.91+2.5.0-22d4614ddf83988a3771fb379ea029e663b4455a-1-cb5e816fa6b1a6b5342df69755869d71-0
openvswitch - 2.5.0-22d4614ddf83988a3771fb379ea029e663b4455a-1
openvswitch-base - 2.5.0-22d4614ddf83988a3771fb379ea029e663b4455a-1
openvswitch-benchmark - 2.5.0-22d4614ddf83988a3771fb379ea029e663b4455a-1
openvswitch-ipsec - 2.5.0-22d4614ddf83988a3771fb379ea029e663b4455a-1
openvswitch-ovs-appctl - 2.5.0-22d4614ddf83988a3771fb379ea029e663b4455a-1
openvswitch-ovs-dpctl - 2.5.0-22d4614ddf83988a3771fb379ea029e663b4455a-1
openvswitch-ovs-ofctl - 2.5.0-22d4614ddf83988a3771fb379ea029e663b4455a-1
openvswitch-ovs-vsctl - 2.5.0-22d4614ddf83988a3771fb379ea029e663b4455a-1
openvswitch-ovsdb-client - 2.5.0-22d4614ddf83988a3771fb379ea029e663b4455a-1
openvswitch-python - 2.5.0-22d4614ddf83988a3771fb379ea029e663b4455a-1

```
Module get installed property
```
root@cpe5:~# dmesg|grep openvswitch
[   14.260253] openvswitch: Open vSwitch switching datapath 2.5.0
[   14.266167] openvswitch: LISP tunneling driver
[   14.270632] openvswitch: STT tunneling driver
```

But, when I try to create a tunnel...
```
ovs-vsctl add-br sw1
ovs-vsctl add-port sw1 lisp0  -- set Interface lisp0 type=lisp options:remote_ip=flow options:key=flow
```
it fails for all typical tunelling protocols.
```
2017-10-30T18:42:37.530Z|00612|dpif|WARN|system@ovs-system: failed to add gre0 as port: Address family not supported by protocol
2017-10-30T18:48:10.813Z|01372|dpif|WARN|system@ovs-system: failed to add lisp0 as port: Address family not supported by protocol
2017-10-30T18:48:10.814Z|01373|dpif|WARN|system@ovs-system: failed to add gre0 as port: Address family not supported by protocol
```

Do you have any idea about what i am doing wrong?
(I use OVS in Ubuntu regularly, but haven´t been able to make it work in Ombia Turris)

Any guideline, will be greatly appreciated!!

jmguzmanc, 2017-11-01:
I am suspecting this is because stable-3.8.5 uses kernel 4.4 and openvswitch 2.5 is supported up to kernel 4.3. 
I will open another issue to ask for help for how to make openvswitch 2.7 work...

ja-pa, 2018-11-09:
Moved to https://gitlab.labs.nic.cz/turris/openwrt/issues/135

closed 2018-11-09 by ja-pa