CZ-NIC/turris-os #70: [Omnia] Set default CPU governor to ondemand

rsalvaterra, 2017-11-10:
According to sysfs, the Armada 385 supports both 800 MHz and 1.6 GHz:

root@turris:~# cat /sys/devices/system/cpu/cpufreq/policy0/scaling_available_frequencies 
800000 1600000

If allowed to scale down, the CPU stays most of the time at 800 MHz:

root@turris:~# cat /sys/devices/system/cpu/cpufreq/policy0/stats/time_in_state 
800000 93758
1600000 3226

For this reason, I'm led to believe the default governor should be ondemand, in order to save power.

BKPepe, 2017-11-18:
Hi!

I think you're looking for official respond from Turris team, which you can find here:
https://forum.turris.cz/t/cpu-scaling-and-cpufreq/2097/4?u=pepe

ja-pa, 2017-11-20:
Just for the record part of explanation from Tomas Hlavacek:
https://forum.turris.cz/t/cpu-scaling-and-cpufreq/2097/4?u=pepe
> I have actually done that on summer in order to test power and thermal profile. Patches with few additional fixes for the Omnia board are here; https://github.com/tmshlvck/omnia-linux/tree/omnia-cpuclk15 .
> 
> Problem is, that it is considered unstable/dangerous feature and it didn't reduce the overall power and thermal profile so much, to justify testing and stabilization effort needed in order to put it into mainline or/and TurrisOS. If somebody have time to test it extensively and possibly cleanup the work, I will be glad to re-send Gregory's patches along with my fixes.
> 
> Another story is CPU idle support, which is also missing and it seems to be result of a HW bug somewhere in I/O subsystem and that has been temporarily resolved (quite a long time ago) by disabling the CPU idle in this patch https://patchwork.kernel.org/patch/6319581/15 . We are investigating this issue in cooperation with Marvell, however I can't make any guarantees or even assumptions o when or if ever it is going to be resolved without disabling the driver.
> 
> However it seems that even with CPU idle and with on-demand cpufreq driver it won't make that big difference. Our experiments indicate that the difference between full power 1600MHz with 100% CPU utilization and idle at 800MHz was only 1-1.5 watts.

closed 2017-11-20 by ja-pa