CZ-NIC/turris-os #71: Broken docker build

petrkr, 2017-11-25:
When trying do toolchain in docker by provided docker files, it will crash on build toolchain docker image with this error:

Build dependency: Please install Git (git-core) >= 1.6.5

run by
```
docker build -t turris/turris-os-build:omnia-toolchain -f toolchain --build-arg FLAVOR=omnia .
```

all log:
```
+ scripts/override_release_numbers.sh                                                                                                        
+ '[' -z '' ']'                                                                                                                              
+ '[' -n '' ']'                                                                                                                              
+ REAL_PKG_BRANCH=                                                                                                                           
+ cat configs/common configs/omnia                                                                                                           
+ sed -e 's|@BOARD@|omnia|' -e 's|@BRANCH@||'                                                                                                
+ '[' -n '' ']'                                                                                                                              
+ echo CONFIG_ALL_KMODS=y                                                                                                                    
+ '[' -n '' ']'                                                                                                                              
+ '[' -f /home/build/mime.key ']'                                                                                                            
+ make_defconfig                                                                                                                             
+ make defconfig                                                                                                                             
Checking 'working-make'... ok.
Checking 'case-sensitive-fs'... ok.
Checking 'gcc'... ok.
Checking 'working-gcc'... ok.
Checking 'g++'... ok.
Checking 'working-g++'... ok.
Checking 'ncurses'... ok.
Checking 'zlib'... ok.
Checking 'libssl'... ok.
Checking 'tar'... ok.
Checking 'find'... ok.
Checking 'bash'... ok.
Checking 'patch'... ok.
Checking 'diff'... ok.
Checking 'cp'... ok.
Checking 'seq'... ok.
Checking 'awk'... ok.
Checking 'grep'... ok.
Checking 'getopt'... ok.
Checking 'stat'... ok.
Checking 'md5sum'... ok.
Checking 'unzip'... ok.
Checking 'bzip2'... ok.
Checking 'wget'... ok.
Checking 'perl'... ok.
Checking 'python'... ok.
Checking 'svn'... ok.
Checking 'git'... failed.
Checking 'file'... ok.
Checking 'openssl'... ok.
Checking 'ldconfig-stub'... ok.

Build dependency: Please install Git (git-core) >= 1.6.5

Prerequisite check failed. Use FORCE=1 to override.
make: *** [staging_dir/host/.prereq-build] Error 1
The command '/bin/sh -c PRIMARY_ONLY=1 ./compile_fw $FLAVOR -j1 tools/install toolchain/install IS_TTY=1 LOGFILE=1 BUILD_LOG=1 V=s $EXTRA_PARAMS' returned a non-zero code: 2
```

Cynerd, 2018-02-26:
This is happening because OpenWRT toolchain doesn't expect git version 2.0. At least it did not supported that at time we fork current version of Turris OS. But base container image is based on Trusty containing version 1.9.1-1. Are you using our Docker file? Haven't you modified base image?

MarianneH93, 2018-03-27:
I am also having this issue. I am using a virtual machine with a fresh install of Debian 9. I haven't modified anything and was following instructions posted [here](https://doc.turris.cz/doc/en/howto/turris_os_build).

sukovec, 2018-04-09:
Hello. All I can say is "#meetoo". 

Tried both testing and stable branches with exactly same result - and the only thing I've changed is commenting last two lines (I need only omnia, not turris) in build-all script. 

As a host, I use Archlinux (4.15.15-1-zen kernel) with Docker version 18.03.0-ce, build 0520e24 (not the one in the Archlinux repos but rather binary stable version from Docker's official website - specifically made by this package: https://aur.archlinux.org/packages/docker-stable-bin/ )

Does someone solved this issue somehow?

Cynerd, 2018-04-11:
Ok as it seems Ubuntu changed some dependencies for some reason. Probably because it's lts so why not change some dependencies. ssh and man was missing. They were added to base image. Also it shouldn't be now pulling in xorg.
See: https://gitlab.labs.nic.cz/turris/openwrt/commit/544aa223c7c08d2acaa17017c91d6f58c35afda3
I tested it and this git error is no longer there.

closed 2018-04-11 by VojtechMyslivec

sukovec, 2018-04-13:
Hello. The build proceeded much further, but still, after several hours (~5) it ended with error. I'm not sure, if I should open new issue or just reopen this, so I will continue, as the problem is still "broken docker build".

This time, it crashed when building LuaJIT, relevant output's here (I've stripped down amalgation warning)

    make -C src amalg
    make[5]: Entering directory `/home/build/openwrt/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.15_eabi/LuaJIT-2.0.5/src'
    make all "LJCORE_O=ljamalg.o"
    make[6]: Entering directory `/home/build/openwrt/build_dir/target-arm_cortex-a9+vfpv3_musl-1.1.15_eabi/LuaJIT-2.0.5/src'
    HOSTCC    host/minilua.o
    In file included from /usr/include/bits/errno.h:24:0,
                    from /usr/include/errno.h:35,
                     from host/minilua.c:40:
    /usr/include/linux/errno.h:1:23: fatal error: asm/errno.h: No such file or directory
     #include <asm/errno.h>
                           ^
    compilation terminated.

Cynerd, 2018-04-13:
There was missing linux-headers package from container. I added it so please try again.

sukovec, 2018-04-17:
Tried it today - exited with the same error.

Cynerd, 2018-04-17:
I know. I tried it yesterday. I am sorry for not letting you know.
For some reason LuaJIT is compiled with -m32 option for host. So it requires 32bit libc. I added it but it failed later on with some linkage error. Probably something is still missing. But I had no time since then to look what packages should be installed to make it work. Feel free to test it.  You should have omnia-toolchain image and you can run it and use `make package/luajit/install`. That should replicate problem with just LuaJIT package very quickly.

sukovec, 2018-04-23:
Thanks for reply - what's the preferred way for adding libraries for the build? I tried to install libc6-dev-x32 (and few other packages) via apt-get, but it didn't help.

Cynerd, 2018-04-23:
In Debian you probably have to enable i386 architecture and install packages from those repositories. But I probably managed to make it run even without that. Installing gcc-multilib should be enough to make luajit to compile. But I ended up with not enough space in docker (for some reason there is some limit on btrfs subvolume in docker although I have quotas disabled). But if you pull openwrt repository now then you should be able to build new image that should work.

sukovec, 2018-04-24:
You are repairing bugs faster, than I'm trying builds. However, this time build finished completely. So thank you very much for your support!