CZ-NIC/turris-os #73: RAID disk do not spin down

stefan-sedlak, 2017-12-16:
with last update 3.9 there is every 5 minutes an activity with my RAID drive
md0 is mounted as /srv
```
root@nas:~# dmesg | grep raid
[ 2426.353839] md0_raid1(1920): WRITE block 8 on sda1 (1 sectors)
[ 2426.353856] md0_raid1(1920): WRITE block 8 on sdb1 (1 sectors)
[ 2726.362546] md0_raid1(1920): WRITE block 8 on sda1 (1 sectors)
[ 2726.362575] md0_raid1(1920): WRITE block 8 on sdb1 (1 sectors)
[ 3026.360764] md0_raid1(1920): WRITE block 8 on sda1 (1 sectors)
[ 3026.360783] md0_raid1(1920): WRITE block 8 on sdb1 (1 sectors)
```

stefan-sedlak, 2017-12-19:
I provided factory reset and it was auto updated to 3.9.

I see in log reading every 1s:
```
[  707.255860] mountd(2137): READ block 0 on sdb1 (32 sectors)
[  707.258279] mountd(2137): READ block 0 on sda1 (32 sectors)
[  708.260652] mountd(2137): READ block 0 on sdb1 (32 sectors)
[  708.262871] mountd(2137): READ block 0 on sda1 (32 sectors)
[  709.265101] mountd(2137): READ block 0 on sdb1 (32 sectors)
[  709.267439] mountd(2137): READ block 0 on sda1 (32 sectors)
```

ja-pa, 2017-12-19:
Hi,
5 minute period is very suspicious. Our watchdog is set to 5 min so it could be related to unbound/kresd or collectd.

Could you enable block_dump ? 
echo 1 > /proc/sys/vm/block_dump

This will log disk writeout to syslog.

cat /tmp/log/messages | grep "debug kernel"

stefan-sedlak, 2017-12-19:
The logs were provided exactly by enablyng  block_dump.

Unfortunatelly I provided factory reset it was really anoying and my wife had no mercy :)
After factory reset and (auto)update to 3.9 mountd was reading every 1s from both unmounted disks...
I killed mountd and hdds were able to sleep and my wife is happy again.

Sry for story, obviously I will not be able provide more info for 1st issue.

stefan-sedlak, 2017-12-22:
when I assembled raid by
`mdadm --assemble /dev/md0 /dev/sda1 /dev/sdb1`
`mountd` started behave normally.

At the moment my raid disks are fine (able to sleep)
I do not know what caused the issue after update to 3.9
However I cannot provide more info so I close the issue.

closed 2017-12-22 by stefan-sedlak