CZ-NIC/turris-os #75: When manually configuring IPv4, Turris should support CIDR in web interface

marek-stopka, 2018-02-01:
When you are manually configuring IPv4 connectivity using web interface (I just opened the package 👍) on Turris Omnia; you need to enter netmask, I have a /29 assigned from ISP and I tried to enter /29 in both netmask and IPv4 address field. None of these have worked, not many people who do not work primarily in networking remember CIDR to NETMASK conversions, it would be more convenient to have the ability to enter CIDR notation IP address in the IP field, after which the NETMASK should either fill itself,disappear or become disabled.

bernrad, 2018-09-27:
I am out of the office until 30/09/2018.

I will respond to your message when I return.

Note: This is an automated response to your message  "Re:
[CZ-NIC/turris-os] When manually configuring IPv4, Turris should support
CIDR in web interface (#75)" sent on 27/09/2018 16:25:48.

This is the only notification you will receive while this person is away.

bernrad, 2018-09-27:
I am out of the office until 30/09/2018.

I will respond to your message when I return.

Note: This is an automated response to your message  "Re:
[CZ-NIC/turris-os] When manually configuring IPv4, Turris should support
CIDR in web interface (#75)" sent on 27/09/2018 16:23:25.

This is the only notification you will receive while this person is away.