CZ-NIC/turris-os #77: Two general questions about turris os...

FrostKnight, 2018-03-29:
One, is the turris os, have any proprietary blobs, firmware, etc,
Basically is it free software?

Two, is the hardware its for, libre?
Are you trying to get RYF certification from gnu? 

I just would like to know before I buy the turris mox or any turris os product.

bernrad, 2018-03-29:
I am out of the office until 06/04/2018.

I will respond to your message when I return.

Note: This is an automated response to your message  "[CZ-NIC/turris-os]
Two general questions about turris os... (#77)" sent on 29/03/2018 02:49:10
.

This is the only notification you will receive while this person is away.

ja-pa, 2018-03-29:
Hi,
I'm aware only of ath10k firmware binary blob.

Source code is available on gitlab and if it's not explicitely declared, it should be provided under the terms of GPLv2 https://gitlab.labs.nic.cz/turris/ (github s mirroring just few repositories from gitlab)

When it comes to RYF certification , it is not in our short term plan (mox campain) as far as I know.

HW documentation:  
Turris Omnia schema documentation - https://doc.turris.cz/doc/_media/rtrom01-schema.pdf

Turris 1.x HW documentation - https://project.turris.cz/en/hardware-documentation

You can also ask on forum.turris.cz which is probably more suitable for this questions.

closed 2018-03-29 by ja-pa

bernrad, 2018-03-29:
I am out of the office until 06/04/2018.

I will respond to your message when I return.

Note: This is an automated response to your message  "Re:
[CZ-NIC/turris-os] Two general questions about turris os... (#77)" sent on
29/03/2018 14:21:31.

This is the only notification you will receive while this person is away.