CZ-NIC/turris-os #79: make menuconfig overrides libc

Shulyaka, 2018-04-12:
Hi,

I try to follow the instructions from https://doc.turris.cz/doc/en/howto/turris_os_build to compile the Turris OS (well, my main purpose is to build a toolchain to compile some updated packages).
I follow these steps:
1. clone the repo
2. `./scripts/feeds update -a`
3. `./scripts/feeds update -a `
4. `cp configs/omnia .config`
5. `make menuconfig` or `make`, which will invoke menuconfig anyway.

After the last step above, the `.config ` file changes, I can see that CONFIG_LIBC_USE_MUSL is gone, and CONFIG_LIBC="uClibc", among other changes.
All binaries built are not runnable on the device. And I cannot build anything without invoking menuconfig.

Best regards,
Denis Shulyaka

Cynerd, 2018-09-27:
You are missing common section of configuration. You have add that one too.

I would suggest you to follow documentation you specified and used `compile_fw` instead. (It was probably updated between your issue creation and me responding to you)

Also you can try to use 
https://repo.turris.cz/omnia/OpenWrt-SDK-mvebu_gcc-4.8-linaro_musl-1.1.15_eabi.Linux-x86_64.tar.bz2

closed 2018-09-27 by VojtechMyslivec

bernrad, 2018-09-27:
I am out of the office until 30/09/2018.

I will respond to your message when I return.

Note: This is an automated response to your message  "Re:
[CZ-NIC/turris-os] make menuconfig overrides libc to uClibc from musl
(#79)" sent on 27/09/2018 15:56:28.

This is the only notification you will receive while this person is away.

bernrad, 2018-09-27:
I am out of the office until 30/09/2018.

I will respond to your message when I return.

Note: This is an automated response to your message  "Re:
[CZ-NIC/turris-os] make menuconfig overrides libc to uClibc from musl
(#79)" sent on 27/09/2018 15:56:50.

This is the only notification you will receive while this person is away.