CZ-NIC/turris-os #80: Build fails on Debian 9

bogdanr, 2018-05-15:
I am trying to build the firmware on Debian 9 and it fails with this message:

```
...
Output of:
arm-openwrt-linux-uclibcgnueabi-gcc -Wall -Wshadow -Wwrite-strings -Wundef -Wstrict-prototypes -Wunused -Wunused-parameter -Wunused-function -Wunused-value -Wmissing-prototypes -Wmissing-declarations -Wno-format-security -Wdeclaration-after-statement -Wold-style-definition -fno-builtin-strlen -finline-limit=0 -fomit-frame-pointer -ffunction-sections -fdata-sections -fno-guess-branch-probability -funsigned-char -falign-functions=1 -falign-jumps=1 -falign-labels=1 -falign-loops=1 -fno-unwind-tables -fno-asynchronous-unwind-tables -fno-builtin-printf -Os -static -o busybox_unstripped -Wl,--start-group applets/built-in.o archival/lib.a archival/libarchive/lib.a console-tools/lib.a coreutils/lib.a coreutils/libcoreutils/lib.a debianutils/lib.a e2fsprogs/lib.a editors/lib.a findutils/lib.a init/lib.a libbb/lib.a libpwdgrp/lib.a loginutils/lib.a mailutils/lib.a miscutils/lib.a modutils/lib.a networking/lib.a networking/libiproute/lib.a networking/udhcp/lib.a printutils/lib.a procps/lib.a runit/lib.a selinux/lib.a shell/lib.a sysklogd/lib.a util-linux/lib.a util-linux/volume_id/lib.a archival/built-in.o archival/libarchive/built-in.o console-tools/built-in.o coreutils/built-in.o coreutils/libcoreutils/built-in.o debianutils/built-in.o e2fsprogs/built-in.o editors/built-in.o findutils/built-in.o init/built-in.o libbb/built-in.o libpwdgrp/built-in.o loginutils/built-in.o mailutils/built-in.o miscutils/built-in.o modutils/built-in.o networking/built-in.o networking/libiproute/built-in.o networking/udhcp/built-in.o printutils/built-in.o procps/built-in.o runit/built-in.o selinux/built-in.o shell/built-in.o sysklogd/built-in.o util-linux/built-in.o util-linux/volume_id/built-in.o -Wl,--end-group -Wl,--start-group -Wl,--end-group
==========
/mnt/turris-os/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-4.8-linaro_uClibc-0.9.33.2_eabi/bin/../lib/gcc/arm-openwrt-linux-uclibcgnueabi/4.8.3/../../../../arm-openwrt-linux-uclibcgnueabi/lib/libc.a(sigrestorer.os):(.ARM.exidx+0x0): undefined reference to `__aeabi_unwind_cpp_pr1'
/mnt/turris-os/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-4.8-linaro_uClibc-0.9.33.2_eabi/bin/../lib/gcc/arm-openwrt-linux-uclibcgnueabi/4.8.3/../../../../arm-openwrt-linux-uclibcgnueabi/lib/libc.a(wait.os):(.ARM.exidx.text.__libc_wait+0x0): undefined reference to `__aeabi_unwind_cpp_pr1'
/mnt/turris-os/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-4.8-linaro_uClibc-0.9.33.2_eabi/bin/../lib/gcc/arm-openwrt-linux-uclibcgnueabi/4.8.3/../../../../arm-openwrt-linux-uclibcgnueabi/lib/libc.a(waitpid.os):(.ARM.exidx.text.__libc_waitpid+0x0): undefined reference to `__aeabi_unwind_cpp_pr0'
/mnt/turris-os/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-4.8-linaro_uClibc-0.9.33.2_eabi/bin/../lib/gcc/arm-openwrt-linux-uclibcgnueabi/4.8.3/../../../../arm-openwrt-linux-uclibcgnueabi/lib/libc.a(close.os):(.ARM.exidx+0x0): undefined reference to `__aeabi_unwind_cpp_pr0'
/mnt/turris-os/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-4.8-linaro_uClibc-0.9.33.2_eabi/bin/../lib/gcc/arm-openwrt-linux-uclibcgnueabi/4.8.3/../../../../arm-openwrt-linux-uclibcgnueabi/lib/libc.a(open.os):(.ARM.exidx+0x0): undefined reference to `__aeabi_unwind_cpp_pr0'
/mnt/turris-os/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-4.8-linaro_uClibc-0.9.33.2_eabi/bin/../lib/gcc/arm-openwrt-linux-uclibcgnueabi/4.8.3/../../../../arm-openwrt-linux-uclibcgnueabi/lib/libc.a(write.os):(.ARM.exidx+0x0): undefined reference to `__aeabi_unwind_cpp_pr0'
/mnt/turris-os/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-4.8-linaro_uClibc-0.9.33.2_eabi/bin/../lib/gcc/arm-openwrt-linux-uclibcgnueabi/4.8.3/../../../../arm-openwrt-linux-uclibcgnueabi/lib/libc.a(read.os):(.ARM.exidx+0x0): undefined reference to `__aeabi_unwind_cpp_pr0'
/mnt/turris-os/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-4.8-linaro_uClibc-0.9.33.2_eabi/bin/../lib/gcc/arm-openwrt-linux-uclibcgnueabi/4.8.3/../../../../arm-openwrt-linux-uclibcgnueabi/lib/libc.a(libc-cancellation.os):(.ARM.exidx.text.__libc_enable_asynccancel+0x0): undefined reference to `__aeabi_unwind_cpp_pr1'
/mnt/turris-os/staging_dir/toolchain-arm_cortex-a9+vfpv3_gcc-4.8-linaro_uClibc-0.9.33.2_eabi/bin/../lib/gcc/arm-openwrt-linux-uclibcgnueabi/4.8.3/../../../../arm-openwrt-linux-uclibcgnueabi/lib/libc.a(libc-cancellation.os):(.ARM.exidx.text.__libc_disable_asynccancel+0x0): undefined reference to `__aeabi_unwind_cpp_pr0'
collect2: error: ld returned 1 exit status
Note: if build needs additional libraries, put them in CONFIG_EXTRA_LDLIBS.
Example: CONFIG_EXTRA_LDLIBS="pthread dl tirpc audit pam"
Makefile:717: recipe for target 'busybox_unstripped' failed
make[4]: *** [busybox_unstripped] Error 1
make[4]: Leaving directory '/mnt/turris-os/build_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/busybox-1.25.1'
Makefile:174: recipe for target '/mnt/turris-os/build_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/busybox-1.25.1/.built' failed
make[3]: *** [/mnt/turris-os/build_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/busybox-1.25.1/.built] Error 2
make[3]: Leaving directory '/mnt/turris-os/package/utils/busybox'
package/Makefile:187: recipe for target 'package/utils/busybox/compile' failed
make[2]: *** [package/utils/busybox/compile] Error 2
make[2]: Leaving directory '/mnt/turris-os'
package/Makefile:184: recipe for target '/mnt/turris-os/staging_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/stamp/.package_compile' failed
make[1]: *** [/mnt/turris-os/staging_dir/target-arm_cortex-a9+vfpv3_uClibc-0.9.33.2_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/mnt/turris-os'
/mnt/turris-os/include/toplevel.mk:181: recipe for target 'world' failed
make: *** [world] Error 2
```

Cynerd, 2018-09-27:
This is 2015 OpenWRT fork. Only known distribution that has somewhat stable builds of this is Ubuntu Trusty. Anything newer has problem with dependencies. I would suggest you to follow https://doc.turris.cz/doc/en/howto/turris_os_build and use either Vagrant or Docker.

Just to note, there is also reported issue that Docker build does not work, just so you know.

closed 2018-09-27 by VojtechMyslivec

bernrad, 2018-09-27:
I am out of the office until 30/09/2018.

I will respond to your message when I return.

Note: This is an automated response to your message  "Re:
[CZ-NIC/turris-os] Build fails on Debian 9 (#80)" sent on 27/09/2018
15:54:39.

This is the only notification you will receive while this person is away.

bernrad, 2018-09-27:
I am out of the office until 30/09/2018.

I will respond to your message when I return.

Note: This is an automated response to your message  "Re:
[CZ-NIC/turris-os] Build fails on Debian 9 (#80)" sent on 27/09/2018
15:52:46.

This is the only notification you will receive while this person is away.