CZ-NIC/turris-os #81: Add ntfs.c to packages/system/fstools to support NTFS drives in /sbin/block

peci1, 2018-05-31:
Upstream already has the needed changes: https://git.openwrt.org/?p=project/fstools.git;a=commit;h=bdeb95a849913bfe2c0eff5d03857c79a4c7d66d . It'd solve a lot of NTFS-related issues from the forum...

bernrad, 2018-05-31:
I am out of the office until 03/06/2018.

I will respond to your message when I return.

Note: This is an automated response to your message  "[CZ-NIC/turris-os]
Add ntfs.c to packages/system/fstools to support NTFS drives in /sbin/block
(#81)" sent on 31/05/2018 12:42:35.

This is the only notification you will receive while this person is away.

ja-pa, 2018-11-09:
Moved to https://gitlab.labs.nic.cz/turris/openwrt/issues/217

closed 2018-11-09 by ja-pa