CZ-NIC/turris-os #84: Docker build broken again - linking libpython2.7

sukovec, 2018-08-29:
Hello.

Few months ago I had a problem building a turris omnia toolchain & OS for docker and now it happened again. This time it was failing due to linker unable to link (native) libpython. Failing due to linker error while compiling CFFI:

    "/usr/bin/ld: cannot find -lpython2.7"

The simple solution (maybe I should call it workaround :) ) is adding following line to `docker/omnia` dockerfile

    RUN sudo ln -s /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0 /usr/lib/x86_64-linux-gnu/libpython2.7.so

Then whole build success. I don't know, if this is correct place for posting, but if not, redirect me please. Thank you.

bernrad, 2018-08-29:
I am out of the office until 11/09/2018.

I will respond to your message when I return.

Note: This is an automated response to your message  "[CZ-NIC/turris-os]
Docker build broken again - linking libpython2.7 (#84)" sent on 29/08/2018
20:11:27.

This is the only notification you will receive while this person is away.

Cynerd, 2018-09-27:
This is really weird. This looks like more like an Ubuntu package bug more than anything else. I am going to try to compile it again and I will see.

bernrad, 2018-09-27:
I am out of the office until 30/09/2018.

I will respond to your message when I return.

Note: This is an automated response to your message  "Re:
[CZ-NIC/turris-os] Docker build broken again - linking libpython2.7 (#84)"
sent on 27/09/2018 15:39:55.

This is the only notification you will receive while this person is away.

ja-pa, 2018-11-09:
Moved to https://gitlab.labs.nic.cz/turris/openwrt/issues/219

closed 2018-11-09 by ja-pa