CZ-NIC/turris-os #85: fail to install lxc containers

necenzurat, 2018-09-18:
````
root@turris:~# lxc-create -t download -n name_of_lxc_container
Setting up the GPG keyring
Downloading the image index
ERROR: Failed to download http://repo.turris.cz/lxc//meta/1.0/index-system
lxc-create: lxccontainer.c: create_run_template: 1201 container creation template for name_of_lxc_container failed
lxc-create: lxc_create.c: main: 274 Error creating container name_of_lxc_container
root@turris:~#
````

Changing the repo to images.linuxcontainers.org works in showing the distributions, but fails and returns an Invalid signature.

Updating the the /usr/share/lxc/templates/lxc-download with the file from https://github.com/lxc/lxc/blob/master/templates/lxc-download.in gives some small progress, but to no avail.

BKPepe, 2018-09-19:
It's fixed now. Thanks for reporting.

closed 2018-09-19 by ja-pa

omichalek, 2019-01-15:
is the fix release please? I still get the same error with Turris 3.11.2

BKPepe, 2019-01-15:
Hello,

You're responding to issue, which is 4 months ago and yes, we have provided the fix to it. However, it might seem like the same problem, but the issue is different. We have to look at it today and once it would be fixed, I will let you know on the forum, where you have also replied, which won't speed the process of fixing and pushing the fix and it might delay it. Copy-paste responses don't look good and I will need to write something different to you on the forum.

This issue doesn't require to release a new version of Turris OS. 

BTW: I'm locking this conversation here, let's continue on our [forum](https://forum.turris.cz/t/lxc-lxc-create-fails-due-to-missing-ssl/7861/20?u=pepe) as I said.

Best regards,
Pepe