CZ-NIC/turris-os #86: Iftop on Turris needs to be rebuilt due to libpcap upgrade

mark-stopka, 2018-09-21:
Hi,
`iftop` on Turris Omnia needs to be rebuilt agains new `libpcap` as the current iftop is built against 1.8.0 and Turris OS now ships 1.9.0 of `libpcap` so the `iftop` stopped working...

````
root@turris01:~# iftop 
Error loading shared library libpcap.so.1.8: No such file or directory (needed by /usr/bin/iftop)
Error relocating /usr/bin/iftop: pcap_compile: symbol not found
Error relocating /usr/bin/iftop: pcap_geterr: symbol not found
Error relocating /usr/bin/iftop: pcap_setfilter: symbol not found
Error relocating /usr/bin/iftop: pcap_open_live: symbol not found
Error relocating /usr/bin/iftop: pcap_loop: symbol not found
Error relocating /usr/bin/iftop: pcap_datalink: symbol not found
root@turris01:~# opkg info iftop
Package: iftop
Version: 1.0pre2-2
Depends: libc, libpcap, libncurses, libpthread
Status: install user installed
Section: net
Architecture: mvebu
Maintainer: Jo-Philipp Wich <jow@openwrt.org>
Size: 21178
Filename: iftop_1.0pre2-2_mvebu.ipk
Source: package/network/utils/iftop
Description: iftop does for network usage what top(1) does for CPU usage. It
 listens to network traffic on a named interface and displays a 
 table of current bandwidth usage by pairs of hosts. Handy for 
 answering the question 'why is our ADSL link so slow?'.
Installed-Time: 1518412358

root@turris01:~# opkg info libpcap
Package: libpcap
Version: 1.9.0-1
Depends: libc
Status: install user installed
Section: libs
Architecture: mvebu
Maintainer: Felix Fietkau <nbd@nbd.name>
Size: 100531
Filename: libpcap_1.9.0-1_mvebu.ipk
Source: package/libs/libpcap
Description: This package contains a system-independent library for user-level network packet
 capture.
Installed-Time: 1537455619

root@turris01:~# opkg install iftop
Package iftop (1.0pre2-2) installed in root is up to date.
root@turris01:~# 
````

BKPepe, 2018-09-24:
Hello,

Thank you for your report. This would be fixed in Turris OS 3.10.7, which we'll release soon.

In the meantime, when we'll release Turris OS 3.10.7, you can fix it with the following command:
`opkg update && opkg install --force-reinstall iftop`

bernrad, 2018-09-24:
I am out of the office until 30/09/2018.

I will respond to your message when I return.

Note: This is an automated response to your message  "Re:
[CZ-NIC/turris-os] Iftop on Turris needs to be rebuilt due to libpcap
upgrade (#86)" sent on 24/09/2018 16:35:06.

This is the only notification you will receive while this person is away.

BKPepe, 2018-09-27:
Fixed in Turris OS 3.10.7, which we released yesterday.

~ Closing

closed 2018-09-27 by VojtechMyslivec

bernrad, 2018-09-27:
I am out of the office until 30/09/2018.

I will respond to your message when I return.

Note: This is an automated response to your message  "Re:
[CZ-NIC/turris-os] Iftop on Turris needs to be rebuilt due to libpcap
upgrade (#86)" sent on 27/09/2018 13:39:32.

This is the only notification you will receive while this person is away.

bernrad, 2018-09-27:
I am out of the office until 30/09/2018.

I will respond to your message when I return.

Note: This is an automated response to your message  "Re:
[CZ-NIC/turris-os] Iftop on Turris needs to be rebuilt due to libpcap
upgrade (#86)" sent on 27/09/2018 13:37:39.

This is the only notification you will receive while this person is away.